BCN3D/BCN3DSigma-Firmware #7: Ayuda: Extrusor golpea la cama

4truemodels, 2015-09-24:
Saludos,
Disculpad que aproveche este mismo medio, supongo que no es el lugar apropiado y en ese caso agradecería cualquier otro método de contacto para este tipo de cosas...

Sucede que desde la última vez que realicé un "Full Calibration" esta mañana, cada vez que intento imprimir cualquier cosa parece que algo ha cambiado en los primeros pasos, puesto que la Cama o no baja lo suficiente o sube demasiado deprisa, lo que provoca que cuando el Extrusor Izquierdo se dirige al primer punto del recorrido de la impresión, golpea con bastante violencia el cristal, para luego empezar a imprimir con total normalidad.
Llevo unas horas dándole vueltas y no entiendo lo que está sucediendo. Si fuese un problema de calibrado, se habría resuelto tras alguno de los siguientes procesos de "Full Calibration" que he llevado a cabo con posterioridad. Además tras ese golpe durante el recorrido desde el "área de reposo" hasta el punto inicial de impresión, el proceso sigue con normalidad y las impresiones terminan correctamente, sin rozar el cristal ni nada por el estilo.
No veo por ninguna parte la causa de que esto me esté sucediendo, puesto que sigo usando por el momento el mismo Cura (el modificado por vosotros) desde el primer día que recibí la máquina, y hasta esta misma mañana no me había sucedido jamás nada parecido.

Os adjunto un link con el "profile" que llevo días usando con el Cura sin ningún problema, y que es con el mismo con el que estoy teniendo estos "sustos" desde esta mañana:
https://drive.google.com/file/d/0BwcplIMBp-swbWt4RG5lY3BpUlE/view?usp=sharing

Espero que podáis echarme un cable y decirme algo pronto, seguiré dándole vueltas y volveré a calibrar la máquina alguna que otra vez más, a ver si termino resolviendo por qué está sucediendo esto.

Muchas gracias!

4truemodels, 2015-09-25:
Os añado información...

Parece que para poder imprimir con esa resolución tan alta, 0'05mm por capa, resulta necesario realizar un "Full Calibration" nuevo, pero en el paso de la calibración del eje Z, en lugar de confirmar el paso una vez el Extrusor roza el folio, lo confirmo justo un paso de motor antes (un poco más alejado de la cama). De este modo parece que me permite imprimir objetos con una capa inicial de 0'1mm (y subsiguientes de 0'05mm) sin dar esos golpes al cristal cada vez que empieza a imprimir.
Desgraciadamente, esto implica que la primera capa no la completa correctamente, dejando algunos huecos entre las líneas del filamento, aunque en las capas posteriores se corrige y cierra por completo el modelo a imprimir.

No sé si existe alguna otra solución a este problema, ni tampoco tengo claro a qué es debido, seguiré dándole algunas vueltas, porque no me parece una solución apropiada perder calidad en la primera capa de impresión.

Un saludo!

flandaluce, 2015-12-18:
es posible que el problema venga dado por estar utilizando una versión del Cura antigua.

En la última que tenemos (0.1.2) ya se modificaron las configuraciones para evitar esta colisión.

;Sliced at: {day} {date} {time}
;Basic settings: Layer height: {layer_height} Walls: {wall_thickness} Fill: {fill_density}
;Print time: {print_time}
;Filament used: {filament_amount}m {filament_weight}g
;Filament cost: {filament_cost}
;M190 S{print_bed_temperature} ;Uncomment to add your own bed temperature line
;M109 S{print_temperature} ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 Z5 F200		 ;extruder/bed crash corrected ******
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Printing...

La linea clave que evita la colisión (G1 Z5 F200). 
Prueba a desinstalar el Cura de tu PC y instalar la versión 0.1.2

kitusmark, 2016-03-09:
Hola @4truemodels @flandaluce 

Si el problema se ha resuelto cargando la nueva versión de firmware, por favor cerrad el "issue"

Muchas gracias.

closed 2016-07-11 by AlejandroGarcia92