BCN3D/BCN3DSigma-Firmware #8: Impresiones "sucias"

4truemodels, 2015-09-25:
¡Hola de nuevo!

Todavía sigo experimentando con la máquina, trabajando a distintas escalas, formas, con un solo extrusor, con los dos, combinando materiales (PLA + PVA), etc...
Por el momento no logro afinar con la Retracción para imprimir, ni siquiera usando vuestros GCodes como configuraciones a emular en el Cura para imprimir mis propios modelos 3D.
El problema es muy grande cuando intento imprimir más de un objeto a la vez, y un inconveniente de menor escala al trabajar con los dos extrusores.
Se trata de que las impresiones salen "sucias", y con esto quiero decir que ambos extrusores generan muchos "hilos perdidos", y es que cuando se mueven sin imprimir todavía sueltan demasiado filamento, generando muchas "rebabas" e hilos finos que ensucian mucho las impresiones y requieren demasiado trabajo de "post-producción" para dejarlas bien.
Por ahora éste problema está resultando bastante molesto, y no termino de encontrarle el truco.
¿Os sucede también a vosotros?

Creo que tiene mucho que ver con haber cambiado el sistema de Extrusión, tirando de tubos de alimentación en lugar del "Direct Extruder" que utiliza la BCN3D+. Lo digo porque es algo que también experimenté en mis inicios en este mundo con la primera versión de la Ultimaker hace 3 años, que utiliza un sistema de alimentación muy similar al de la SIGMA.
¿Alguna sugerencia para reducir al máximo este inconveniente?

¡Gracias por vuestro trabajo!

PD: mi última prueba sin éxito ha sido retocando la Distancia de Retracción en el Cura, incrementándola a 6'0mm, pero dejando intacta su Velocidad en 45mm/s.

4truemodels, 2015-09-28:
¡Saludos!

Definitivamente he encontrado la causa a estas impresiones "sucias". Mis sospechas sobre el tipo de extrusor que lleva instalados la SIGMA eran infundadas, toda la responsabilidad recae en el material de impresión, en este caso en el PLA de Colorfila.

He recibido una muestra gratuita de ABS de una empresa llamada SmartFill, y usando su material con la configuración por defecto del Cura sobre Retracción (Velocidad y Distancia) no ha generado ni una sola rebaba ni hilos entre los distintos objetos durante su impresión.
No me lo esperaba, así que esto me llevará a probar otros materiales, como vuestro ABS Premium y PLAs de otras marcas, en busca de aquellos que sean mucho más "agradecidos" a la hora de imprimir.

Por el momento estoy más que decepcionado con el material de Colorfila, con las BCN3D+ todavía puede uno pensárselo y usarlos, pero desde luego no con la SIGMA, porque su "suciedad" y sobretodo sus constantes roturas durante la introducción o retirada del filamento lo hacen 100% aborrecible como material de trabajo. Lamento tener que dar una opinión tan negativa sobre este material, pero no puedo recomendar su uso a nadie y me siento prácticamente "estafado" tras haber adquirido unas cuantas bobinas de ese material más caro que el PLA "de baja calidad".

Sea como sea, sigo opinando que vuestro trabajo diseñando y construyendo la SIGMA me parece soberbio, aunque siempre queden flecos y espacio para las mejoras, mi opinión sobre esta nueva impresora es muy buena.

PD: ¿qué sucede con los modelos 3D de las partes impresas de la máquina? Solicité algunos para poder fabricarme los recambios que necesito y sigo sin respuesta, tanto por este medio como a través del correo electrónico.

neildarlow, 2015-11-21:
I too noticed fine threads on prints. Increasing the retraction to 4.5mm eliminated them for me.

Also, there is a retraction setting in Cura (Minimal extrusion before retracting - in Expert settings) which if not set to zero can cause excessive threads between multiple objects and parts of single objects.

closed 2016-07-11 by AlejandroGarcia92