BCN3D/BCN3DSigma-Firmware #27: Left extruder Bed Calibration probe position problem.

neildarlow, 2015-11-21:
The glass platform has a 45 degree bevel (or chamfer if you prefer that term).

I am finding that during Bed Calibration the left extruder microswitch is making contact with the glass on the bevel and not on the flat glass surface.

This, I believe, can introduce an error into the bed calibration that raises the left side of the bed and could cause the nozzle to hit the surface as noticed in another issue here.

xawox, 2015-11-30:
In theory the detection position for bed calibration isn't this position. 
Which version of firmware have you? (It should be displayed in the logo screen, but if it not appears should be v01-0.0.5)

neildarlow, 2015-11-30:
I will confirm that with Pronterface. The logo screen is not displayed long enough for me to read the firmware version.

neildarlow, 2015-11-30:
Yes. My firmware version is 01-0.0.5. Confirmed with Pronterface.

xawox, 2015-11-30:
Try to update the firmware with the v01-1.0.0 release, to check if it's solves this problem.

It include the material selection, but it's still in improving process. Sooner we'll publish a new release than includes improves and corrections

https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Updating-the-BCN3D-Electronics-firmware

This link shows how to update firmware with Cura from  BCN3D. Just in case, after update the firmware, use pronterface to reset the EEPROM memory (command M502 to reset EEPROM and M500 to save the reset).

neildarlow, 2015-11-30:
Two problems:

1) The left probing position is definitely off. The carriage puts enough pressure on the bevel of the glass that you can see the right side of the glass plate lift up. That is a hell of a lot of pressure!

2) I have BuildTak on my bed. During Z Calibration the nozzle is now pushed hard into the BuildTak and there is not enough upward movement permitted to allow the nozzle to clear the BuildTak.

I was forced to abort the Z Calibration and the nozzle is now embedded in the BuildTak. I need to manually lower the Z axis but I dare not perform a G28 (Home) in Pronterface because it will probably tear the bed assembly apart!

closed 2016-07-11 by AlejandroGarcia92