BCN3D/BCN3DSigma-Firmware #28: Z-calibration part of Full Calibration requires modification

neildarlow, 2015-11-22:
During Full Calibration the Z height is set in a procedure as follows:
1) Heat both extruders
2) Permit user to wipe left extruder
3) Permit user to wipe right extruder
4) Perform left extruder Z-calibration
5) Perform right extruder Z-calibration

The left extruder Z-calibration can take some time while the user adjusts the nozzle height repeatedly and during this time the right extruder is oozing plastic. When the time arrives for the right extruder Z-calibration there is significant build-up of plastic on the nozzle which should be wiped away.

I suggest a better procedure would be:
For each extruder --
1) Heat extruder
2) Permit user to wipe this extruder
3) Perform this extruder Z-calibration
Repeat for other extruder

By heating the extruder just before performing the wipe and Z-calibration the nozzle is given very little time to ooze plastic before the calibration sequence is performed.

xawox, 2015-11-30:
We just implement these changes in the new firmware versions. In the v01-1.0.0 it does:
0) Meanwhile the bed calibration works, it heats both extruders to clean temperature.
1) Allow user to wipe left extruder
2) Allow user to wipe right extruder
3) Calibrate Z left extruder
4) Heats left extruder to print temperature
5) Test print for ensure best calibration for this extruder
Repeat procedure for other extruder since step 3.

In the development firmware the procedure is:
0) Meanwhile the bed calibration, it heats both extruders to clean temperature.
1) Allow user to wipe left extruder
2) Calibrate this extruder
3) Heats left extruder to print temperature
3) Test print for ensure best calibration for this extruder
Repeat procedure for other extruder since step 1.

neildarlow, 2015-11-30:
That's good, thank you.

closed 2016-07-11 by AlejandroGarcia92