BCN3D/BCN3DSigma-Firmware #29: Bed temperature for Full Calibration needs review

neildarlow, 2015-11-22:
There is a significant temperature drop between the heated bed PCB and the top surface of the glass. The PLA pre-heat temperatures specified in the firmware appear to be 45C or 50C which although near to the Tg of PLA does not account for the temperature drop across the glass.

I calculated that to achieve a 55C glass surface temperature (which I use on other printers when printing PLA) the heated bed needs to be set to 85C. I did this as an experiment and measured the glass surface temperature to be 56C which confirms my calculation.

By having the lower temperature the glass surface is not sufficiently hot enough to permit PLA to stick reliably during Full Calibration.

Thinking further on this, is there a reason why the Full Calibration alignment test cannot be supplied as a STL or G-Code outside the firmware? If that were possible, users could set bed temperature to suit the materials they are using instead of assuming that PLA is being used.

xawox, 2015-11-30:
We are working on that. We are measuring the difference between hotbed and the surface to add an offset from firmware.

To solve problems with differents materials in the full calibration, we implement  the selection of material when user does insert filament. It allows set the print, insert and remove temperatures and the surface temperature also.

We are also working on define new calibration gcodes for calibrate the axis per separate and control from pronterface or some software serie control. We take account your advise to implement temperatures control for these calibration gcodes

neildarlow, 2015-11-30:
I do not know if offsetting in firmware is a good idea. Think about the situation where a user may have e.g. BuildTak on the bed which will introduce additional offset.

The user can compensate all of this within the temperatures specified in the slicer.

The problem is for the calibration print. If the user is able to specify temperatures for this by some means then that may be a solution.

closed 2016-07-11 by AlejandroGarcia92