BCN3D/BCN3DSigma-Firmware #30: Preheat starts as soon as screen is entered.

neildarlow, 2015-11-27:
When performing Preheat the temperatures start increasing as soon as the function is selected not when Accept is pressed.

This means that if you wanted to set a temperature of e.g. 85C for a PLA cold-pull by the time you have adjusted the temperature down to that value the extruder might be hotter than 85C.

Worse still is that there is no opportunity to reduce temperatures of items you do not want to preheat before Accepting. Both extruders and bed are heated on entry which is not sensible.

xawox, 2015-11-30:
We decided to implement this procedure to reduce the heating time.
Now we are working to heat extruders and bed at print material temperatures and not to 215ºC and 50ºC as default values

neildarlow, 2015-11-30:
This does not address the problem of only heating selected elements. What if, for example, the right extruder is left empty for a while? You do not want to bake plastic residue into the hotend by heating it unnecessarily.

I think it is better to not heat anything until the user has had an opportunity to adjust the preheat temperatures. If he is happy with the temperatures that are presented to him all he has to do is press the Accept icon immediately. It will not delay preheat significantly.

Zorgonaute84, 2016-02-01:
And to redude the cowling time (and the noise) is it possible to switch on the nozzle fan only when the nozzle is heating up ? It could be very usefull when the print is finish to shut off the noisy fan.

closed 2016-07-11 by AlejandroGarcia92