BCN3D/BCN3DSigma-Firmware #31: Why all the adjustment of current_position[Z_AXIS] during extruder calibration?

neildarlow, 2015-11-30:
In Marlin_main.cpp the code following line 2174 (search "case 40") and line 2405 (search "case 41") has many differing values for current_position[Z_AXIS].

There are static assignments of 0.65, 0.5 and 0.2 and modifications of -=0.2, +=0.2, +=2 and -=2. In other parts I have seen 0.4 statically assigned.

Why is there all this vertical adjustment of the nozzle position during the Calibration pattern printing?

If Z has already been calibrated I would expect that a single layer height would be used for the whole Calibration pattern print - maybe 0.2mm.

xawox, 2015-11-30:
It's fixed in v01-1.0.0. Thank you for the feedback and take a look to the code

neildarlow, 2015-11-30:
Are you sure? I based my observations on the current code in the repository.
If v01-1.0.0 differs from current then there must be some changes that need merging into current.
I think it is worth checking them.

xawox, 2015-11-30:
In the master branch (it's the same than v01-1.0.0) https://github.com/BCN3D/BCN3DSigma-Firmware/blob/master/Marlin/Marlin_main.cpp in G40, it moves the bed until 0.65mm statically for avoid the colision, then it purges and retracts (to ensure to it prints but not fall drops), calculate the diferent offsets, moves Y, then X and it adjusts the height at 0.2mm statically, it purges for recover the retract and prints the skirt and lines. It has an current_position[Z_AXIS] "change" between the skirt and the lines, to play with the lines height (is placed at 0.2mm too) .

It's true than in right_test_print_code there is a mistake, an += instead of =

sorry for the code comments, they aren't the best comments in the world precisely
[image]
[image]

neildarlow, 2015-11-30:
Yes, I can see the need for some raising of the hotend for the second set of lines to clear the first set of lines and their skirt.

I just thought that there was a lot of Z_AXIS manipulation going on and possibly more than necessary. At least the erroneous += was located so that is good.

If I had been doing this chunk of code myself, I would create an STL of the desired pattern and use the generated GCode as a template for coding the solution. All the hard work would be done by the slicer :)

closed 2016-07-11 by AlejandroGarcia92