BCN3D/BCN3DSigma-Firmware #33: [Enhancement] Separate calibration tests

pgilfernandez, 2016-01-19:
Sometimes is useful to run just one calibration test and not all of them (for instance dual extruder alignement).
It would be great to have an option to run all of them at once (already included) and all of them individually (now only bed calibration is available).

Thanks

Zorgonaute84, 2016-02-01:
Totaly agree !!!

I think it's useless to calibrate regulary the aligment between the 2 heads. So it could be very great to be able to calibrage only the Z heigt.

It could be greater again, if it's possible to add a quick manual calibration process. For example:
1/ Unscrew the 2 fronts screws
2/ Calibrate the Z height of the left extruder just upside the back screw (with the calibration sheet and the Touch Screen)
3/ Calibrate the Z height just above the front left screw (with calibration sheet and the screw)
4/ Same thing with the front right screw
5/ Finally calibrate the right extruder in the middle of the plate with the calibration sheet and the touch screen.

In few steps, you've got a plan and perfectly calibrate boths extruders.

Sure it's more manual, but for power user it could be very very very faster !

thegreystone, 2016-03-05:
Yep. Agree! +1

thegreystone, 2016-03-05:
To be specific - I want to be able to run ONLY the alignment between the two heads. Or rather, tune the alignment between the two heads without having to run the Z-calibration.

mrcg, 2016-03-08:
1. Each seprately ... only z-calibration, only x-calibration, only y-calibration (beside full calibration)
OR
2. only z-calibration, onlyx&y-calibration

Z-Calibration is the most important. Usecase: with or without blue tape (or capton tape).
Everything seperately would be really useful beside the full calibration.

closed 2016-07-11 by AlejandroGarcia92