BCN3D/BCN3DSigma-Firmware #34: Misalignments on dual extruder after calibration run

pgilfernandez, 2016-01-19:
I have fine tuned X and Y dual extruder calibration with the "lines test" and I have noticed you don't get a  perfect alignment on X axis: just after the tuning I have printed a dual extruder file, it's just a cube but enough to see that it gets the misalignment... Of course I was able to select another number in the "lines test" in order to compensate the misalignment but I wanted to let you know.
If you want pictures let me know.

pgilfernandez, 2016-01-20:
Forget about the misaligment, after iterative trials and errors I have seen the test has some kind of randomness when drawing the lines (that is, with a specific machine and running several times the calibration tests you won't select the same 1 to 10 numbers on each one) so after running some of them trying to select the best aligned lines on X and Y and printing a dual color probe I was able to arrive to a perfect mach.
Summing it up, I think the calibration test sometimes could arrive to a non perfect extruder matching even if you select the best aligned lines.
Please, just test it out

PS - As I have suggested on [this enhancement](https://github.com/BCN3D/BCN3DSigma-Firmware/issues/33) it would be great to have to option to run the calibrations separately. That way it would be really fast to just tweak, for instance, extruder X alignment.

kitusmark, 2016-01-27:
Hi @pgilfernandez. We'll take note on that. Bear in mind that this process is iterative until you get "your" perfect match, just like you did. Tests like this give us a lot of info. 

Thanks!

closed 2016-07-11 by AlejandroGarcia92