BCN3D/BCN3DSigma-Firmware #35: Full calibration crashes

pgilfernandez, 2016-01-20:
After all my tests I have seen there is a problem in the Full calibration routine that sometimes crashes the software and it reboots.
When setting the right extruder Z distance to bed, if you press twice the "OK" button once you have adjusted it, it sometimes crashes and reboots.
I don't know if this is the real problem but it happens to me.

Zorgonaute84, 2016-02-01:
Same issue here. I noticed that the crash looks appeared (on the right extrudeur too) when I click on the DOWN button just before to ACCEPT). But I'm not able to reproduce as I want the issue...

Zorgonaute84, 2016-03-02:
The issue seems appear when you click on the Accept button 2 times quickly.

pgilfernandez, 2016-03-02:
Yes, I have also experienced the same issue

Pablo Gil Fernández
email:
Twitter:
pgilfernandez@gmail.com
@exkr

kitusmark, 2016-03-09:
Hi @pgilfernandez  @Zorgonaute84 

We've detected a couple of places where you can hang up the printer by touching repeatedly. @AlejandroGarcia92 is working on it. In the next firmware version will be fixed. If you detect the same thing in other screens of the display, please post them here to be able to track them .

Thanks.

Zorgonaute84, 2016-03-09:
Ok, That's right I occured the same issue at different place. I'll check for you.

Thanks to hardly work to make this already amazing printer better...

closed 2016-07-11 by AlejandroGarcia92