BCN3D/BCN3DSigma-Firmware #36: Left Z high leveling print test is misaligned

pgilfernandez, 2016-01-20:
You can easily check it on the next picture:
[image]
The red one is the left Z test and from 1 to 5 it shifts to the bottom while the white one, right Z test, is correctly drawn.

kitusmark, 2016-03-09:
Hi @pgilfernandez. Is it fixed with the new firmware Release? v1.1.4? 

If it's fixed please close the issue. 

Thanks.

mrcg, 2016-03-09:
My Sigma is loosing all EEPROM varialbles when switched off for longer time.
All calibration data and filament temperatures are impacted.
When data are lost the 1st full calibration looks also like shown in the picture.
2nd calibration (when data are not lost) calibration pattern are printed correctly.

@pgilfernandez : Do you experience also loss of calibration data from time to time?

If yes please report output from M503 if you still have the issue.

closed 2016-07-11 by AlejandroGarcia92