BCN3D/BCN3DSigma-Firmware #37: "line tests" extruding settings

pgilfernandez, 2016-01-20:
It would be necessary to update the printing settings for the line tests:
- it leaks always filament due to a low retraction applied. It surely be related filament materials but the most common retraction you have set on your profiles is 4mm and may work better
- before starting extruding the tests it purges the nozzle, it eventually happens to me that it is necessary to purge more millimetres because it starts extruding "nothing" (this is due to material dripping when the nozzle is hot and not on use). It is very notable on Z leveling tests because its brim line is so small, maybe, a fast solution would be to increase the brim lines to 2 or 3.

Zorgonaute84, 2016-02-01:
Same issue here. Now, I manually purge the nozzle before lunch any print or full calibration.

mrcg, 2016-03-01:
Same issue to me. If dual print was done before then it could also happen that last dual-retraction was 8mm. In general i would recommend to add small retraction (0.1mm) to end.gcode and end2.gcode (dual) to neutralize the retraction in any case after print has finished.

end.gcode
G1 F50 E0.1

end2.gcode
T0
G1 F50 E0.1
T1
G1 F50 E0.1

Anyhow this feature request still require some more retraction before the line tests starts.
I would like to prime more lines (e.g. box with two lines) rater than priming to the box.

4truemodels, 2016-03-05:
I suffered sometimes, although not many, the same thing, but usually I get that solved by pressing the "Redo" options 1 or 2 times until both extruders recover their flow rate.

closed 2016-07-11 by AlejandroGarcia92