BCN3D/BCN3DSigma-Firmware #38: [Help] understanding touch screen

pgilfernandez, 2016-01-20:
I would like to help on the firmware development, especially on the GUI part but I don't know how the different screens are designed and compiled. I know they are stored in the "SD files v100" but don't know how to edit them... workflow? software?
Any help would be great in order to start helping the development.

Thank you

kitusmark, 2016-01-21:
Hi @pgilfernandez !
We'd love to have all the help we could find. The screen is a module from 4D Systems and they have their own "Arduino" like libraries and their own IDE. It's called Workshop.

All the documentation is on 4D Systems webpage. If you need something more, contact us!

Thanks

pgilfernandez, 2016-01-21:
Hi @kitusmark,

Nice to talk with you and thanks for the info, it made me arrive to the documents and to understand how the GUI is done.
I have also discovered @dryrain repo [RepRapBCN_TouchScreen](https://github.com/dryrain/RepRapBCN_TouchScreen) and played with it... is that your working repo or is it an old one?
I have found lots of things to improve (bad screen resolutions, stretched images, icons to improve...). Could I ask for original design files? are you working with vector or raster designs?

kitusmark, 2016-01-27:
@dryrain was our former developer. That repository must be way way outdated. We 're planning on releasing the original files soon. I'll keep you informed.

pgilfernandez, 2016-01-27:
OK, I'll wait for your advice, thanks

4truemodels, 2016-01-31:
Hi there, I'm pretty interested too in helping with this...
I would like to modify the "SD files" in order to translate all the screen's texts to Spanish and Català.
Please, tell us whenever we can get those files updated to give us the chance of doing so.
Thank you!

pgilfernandez, 2016-01-31:
It would be great to have the original .psd or vectorial file that includes all the screens in order to modify/improve them and/or translate them into Spanish or other languages as @4truemodels suggests.
I guess you will upload it all when you are ready, we will wait for it =)

kitusmark, 2016-02-02:
Let's talk about the LCD screen development. I'm not either the developer or the designer of the screen but I have access to all the files and manage the github account. 

My plan is to open a new repository just for the LCD Screen and put all the files needed there. That way we can work at the same time. Once we have a tested version we can release the files. 

The compiled files then will be available from this repository in order to people get the latest files. 

Traductions and improvements will be in another repository where people can go a download new stuff.

What do you think about it?

4truemodels, 2016-03-05:
Ei, how is it going? Would you open those repositories?

closed 2016-07-11 by AlejandroGarcia92