BCN3D/BCN3DSigma-Firmware #42: [Enhancement] - Manual axis movement

Zorgonaute84, 2016-02-01:
It could be very usefull to have a menu to move the different axis.

- Move X/Y the right and the left extruder (could be with the hands)
- Move Z up and down (couldn't be by hands) VERY VERY VERY USEFULL to do some maintenance on the printer

- Auto-home for both extruders
- Present the left extruder (The left extruder come to the front of the printer and the bed go down.
- Same thing for the right extruder

kitusmark, 2016-02-02:
We've discussed a lot about this but finally we decided not to implement it. 

You can use a computer with pronterface or another serial communication tool to send gcodes to the printer, if that works for you.

Also, we're working on implementing a raspberry Pi with Octoprint for the Sigma in order to be able to print and monitor the printer wireless.

Also I refer you to another issue where some people ofered to help in the development of the LCD #38

closed 2016-02-02 by kitusmark

Zorgonaute84, 2016-02-02:
Oh it's very disappointing to hear that. Please, don't do like big compagnis which believe to know the needs of theirs customers... Listenning the voice of the customer is the key to success...

If I'm the only one who want this feature I understand, but...

4truemodels, 2016-02-03:
I'm with you @Zorgonaute84, I would also love to have access to this kind of services without need to plug in the printer to any computer. In fact, I had already asked to them for this long ago, in a phone call, but I got the same answer as you.

Zorgonaute84, 2016-02-03:
Cool. If more people come here to say that they want this feature, maybe something will happend...

Zorgonaute84, 2016-02-03:
And voila !!!

It just happend the exact reason why you have to implement that.

Don't ask me why, but I move manually the left extruder during the insert filament process. But I did that during the bed is leveling up. So, the bed was move until the extrem up limit, I immediatly stopped the machine. And I'm fucked...

First, you should let the motor block durring this process to avoid my mistake.

Second, how can I do now ? My Macbook with Cura doesn't detect my printer, so I'm f** to move the bed with pronterface too...

4truemodels, 2016-02-03:
Hi @Zorgonaute84,
I don't know anything about Mac, but if you can install both Pronterface and Arduino software you should be able to connect your Sigma and move anything using Pronterface.

Pronterface Mac's versions here: http://koti.kapsi.fi/~kliment/printrun/
Arduino Software here: https://www.arduino.cc/en/Main/Software

You should install Arduino and its drivers (included in the install package) first.
Then plug in your Sigma through any USB port and wait any driver install process (no clue if Mac also does this things).
Once it is done you should run Arduino Software, having your Sigma still plugged in.
Inside Arduino, go to Tools -> Serial Port to check if your Mac really recognizes your Sigma. If the program shows you any port avialable, then you have it! Write the port number it shows you and close Arduino.

Now it is time to open Pronterface, select the same port to connect with your Sigma as you have seen in Arduino (remember you need to close this program before). Wait some secs and you will see how Pronterface shows you a text confirming the correct initiation fo the Sigma (unlocking all command buttons).

I really hope any of these would help you. But yeah, I'm totally agree that BCN3D should insert this options into their Sigma Firmware, in some advance menu or similar. There are many thing that can happen like your accident or even worse.

Zorgonaute84, 2016-02-04:
Hello,

I found where is the problem on my Mac. In the sigma it's not a genuine Arduino card, so I had to install the custom driver. No I can connect it with Cura, but the pronterface in Cura allow to move X and Y axis of the left extruder but not the bed !!!!
 
Hopefully, I succeeed to reajust my bed. I removed the glass plate, run a process, replace the glass and that's ok...

maalencar, 2016-06-05:
Zorgonaute84, could you please share where you found the custom driver for Mac?
I'm having the same problem you had. It's annoying that a printer at this level suffers from such basic functionality flaws.

Anyway, thanks in advance.