BCN3D/BCN3DSigma-Firmware #45: New v01-1.1.3 doesn´t update correctly?

pgilfernandez, 2016-02-02:
I have just updated my Sigma with yesterdays commits and I think it is not correctly updated:
- from Cura (I tried it on OSX) it says there is no new firmware online. I did it by downloading it from Github and then uploading a custom firmware. I of course also copied and overwritten the LCD files
- now the startup screen reads version "v02-1.1.3_d"
- even updating the LCD files to v113 it's not possible to get the supposed new feature _"Calibration prints have been redistributed for better visualization and performace"_
Once found the previously explained issues I haven't continued testing but I would like to know if it has been correctly uploaded or it's me that have done anything wrong...

Thanks

Zorgonaute84, 2016-02-02:
Have you did something to connect Cura (osX) with your sigma ? In my case, Cura want to use the bluetooth connection only... Of course, it didn't find the printer...

kitusmark, 2016-02-02:
Hi @pgilfernandez and @Zorgonaute84.

Yesterday release was a test, we're working in some details to make update easier. The final Release will be really really soon. Justpay atention to the social networks.

Thanks to be so responsive

closed 2016-02-02 by kitusmark

Zorgonaute84, 2016-02-02:
Ok, but is it possible that I already have the 1.1.3 version ? My printer is new, I bought it last week.

kitusmark, 2016-02-02:
@Zorgonaute84 Yes it's possible because one of the first places we introduce the new versions is in production for testing. Anyway when the oficial release come out, it's recommended to update again. All the information will be in the [wiki](https://github.com/BCN3D/BCN3D-Cura-Windows/wiki)