BCN3D/BCN3DSigma-Firmware #46: [Enhancement] LCD screen during printing

Zorgonaute84, 2016-02-02:
It could be very usefull to have a direct look (without clicking on settings) of the target for each value.

Something like:

204°C/205°C         51°C/50°C          206°C/2005°C        100%

4truemodels, 2016-02-03:
Hi there @Zorgonaute84, 
That info you are talking about is already accesible if you press the button to change temperature configuration during the printing, in that screen you can see the temperature objective for both each extruder and the bed.

I talked to them in the past asking for showing something else in the main screen or into another one accesible like the temperatures... the printing time in the LCD! 
I would really love to have that info as the timing calculated by Cura, Slic3r or GCode Viewer are not really so accurate, and when you need to know the exact cost of each printing it is quite annoying.
It should be really good to have the printing time info accesible in the LCD, and to make the printer stop the "clock" once the printing is finished so no matter when you'll come back and check it, you will have access to the exact printing time for each of your works.
Unfortunately that change seems not important for them, as they always answer me with a NO, NO NEED or NO BY NOW.

Zorgonaute84, 2016-02-03:
Yes you right, the information is visible in settings. But I proposed just an improvement to also see all informations (real and target) at the same time.

But as you say, the very usefull feature is to have the real temperature for each extruder and bed on any screen !

closed 2016-07-11 by AlejandroGarcia92