BCN3D/BCN3DSigma-Firmware #47: [Issue] Long time before heating

Zorgonaute84, 2016-02-02:
I noticed that sometime (often) after selected a gcode into the Sd Card, the printer take long time (more than a 1 minute) before heating the extruders. You can show that in the settings where the target still 0 a long time too...

4truemodels, 2016-02-03:
Hi @Zorgonaute84,
I think that long time for the preheating process you are talking about happens because the printer starts heating the platform, and won't start heating any extruder until the platform reaches the temperature needed.
It meens the only time you "lose" is the one needed for the extruder heating after the platform heating so it should be so long as the extruder heats really fast.
Maybe I misunderstood you and you are suffering for some kind of issue I didn't notice by now with my printer. (Hopefully it is not the case).

Zorgonaute84, 2016-02-03:
Ok maybe, I will have a look on that. But it could be improve, because it's very annoying to see the extruders cooldown whereas they already hot... And sometime I sliced my model with a temperature fews days before, and want to change this temperature without have to slice again. I can't do that immediatly. Because when I click to print my model, then settings, the temperature target still 0 for a long time. I'm not sure it's normal.

I will take a video this evening.

4truemodels, 2016-02-03:
You would only be able to change the extruders' temperature once the printing process has really begun.
As the printer starts heating the platform, it even doesn't "read" the proper line of the GCode for the extruders so any change you do "on screen" will be useless as when the bed got heated the printer will "read" that line and will change again the temperature settings as the GCode says.
I had worked with this issue for a long time so that's why I felt free to answer you.

If you want to change the extruders' temperature for the printing you will need to wait until you see the motors begin to move. You will lose the first secons while the printer starts the first layer, but once you set up a new temperature it will be saved and will continue its' work achieving the new temperature and keeping it for the rest of the printing.

Maybe is not a bad idea to change the way the printer does the heating process before printing, but having the nozzles cold while time goes on heatting the platform seems not really a bad idea to me. That makes the process "safer" as the filament won't "over-flow" or even "clog" the nozzle.
It is true that for a PLA printing could be something useless, but when you want to print ABS or other filaments with bed-temperatures of 80ºC or greater... I really think this process is a good idea.

Zorgonaute84, 2016-02-03:
Ok, as you explainend I perfectly understand your point of view. In that case, the good thing is to implement a preheat function for the bed.

So the printing workflow could be:
1/ PreHeat the bed
2/ Slice your model and upload it to your sdcard
3/ Insert the SDCard and launch the print.

4truemodels, 2016-02-03:
You got there a pretty good idea.
In fact I will do so while BCN3D thinks about these posts. 
It will be a little hard to preheat only the bed the way is working now the firmware, but I'm sure the time lost setting temperatures like 0ºC/100ºC/0ºC in the Preheat Screen will be valuable to speed up every printing.
I agree with you and will pray for a "Preheat Bed" function in new firmware updates.

kitusmark, 2016-02-04:
@Zorgonaute84  is like @4truemodels  said. This behavior is default by the original Marlin. At the end, all depends on your workflow. What we do is to turn on the machine, go to preheat menu and select the temperatures we want. Then we slice the model in the computer and store it in the SD card. When we start the print, the temperatures are more or less close to target and we usually wait between 1 to 2 minutes.

It's true it could be a function to only preheat the bed. But the menus tend to get more complex. As you can see, we opted for the simplest but functional interface. This doesn't mean this feedback will arrive to the correct people. :smile: :+1:

closed 2016-07-11 by AlejandroGarcia92