BCN3D/BCN3DSigma-Firmware #48: [Enhancement] Noise reduction

Zorgonaute84, 2016-02-03:
In this ticket I want to share you a tips to reduce the noise of the sigma and an engancement request.

About the request, is it possible to activate the heated head fan only when the nozzle is heating or cooling ? For maintenance, it's intersting to power on the sigma just for activate the light system and the fan noise is anoying and useless.

About the tips. I significantly reduce the noise during printing by glue with a heated glue pistol the edges (inside) of the two container bellow the heads. It's reduce the vibration and the noise.

mrcg, 2016-02-21:
+1

* NTC fan control for the extruder fan. If it's not used it should be off or very low rotation
* Isolate fan with rubber/bumper from metallic housing

The high noise level is really something what needs to be improved on the Sigma.

Zorgonaute84, 2016-02-22:
I significatly reduced the noise of my Sigma with hot glue gun !

First, I harden the two waste container by adding some glue into each edges of the container. It's stopped any vibration...

Secondly I totally disassembled my 2 heads and I had some glue too into each edges of the 2 metal plates (inside)

Then I cuted small circles (2mm) in the cold glue tube, and I inserted them between the 2 feeder motor and the metal frame to reduce the pervasion of the motor vibration into the frame.

And finally I isolated the middle blue plastic part than the others with felt.

pgilfernandez, 2016-02-22:
@Zorgonaute84 could you share some pictures?

Thank you!

Zorgonaute84, 2016-02-23:
Yes of course. I will take some picture this evening.

kitusmark, 2016-02-23:
We'd love to see those pictures too!

Zorgonaute84, 2016-02-24:
So, you can find some pictures of the hack I dit on my Sigma.

You can see the different place where I added some hot glue (no visible too much because the glue is inside) You can see what I cutted to place between the feeder motor and the metal frame, and finally the felt used to isolated the blue part.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pgilfernandez, 2016-02-24:
Thank you very much!
As I see, the noise produced by the rotating fans is not due to the fan case and the metal case, but on the bended parts of the metal hotend housing... is that right?

Zorgonaute84, 2016-02-24:
I can't answer you exactly because I can't heard the sound of your Sigma. I had 2 differents Sigma, and each one produced different sounds. The first one was globally quiter than my actual , but the fans were much noisy. 

The only things that I can tell you with certitude is that most of the noise produced by my actual Sigma was caused by the vibration of the different bendeded parts. So you are right...

pgilfernandez, 2016-02-24:
Ok, thanks for the comment.
I will check mine and try to fight the noise =)

Pablo Gil Fernández
email:
Twitter:
pgilfernandez@gmail.com
@exkr

kapetanikolas, 2016-02-29:
Hi I added a astrosyn nema 17 vibration damper on the Y axis motor and it does make a big difference on the noise. it sounds much more smooth now. Will try it on the Z axis to reduce noise during retraction Z-hop.  I will try to modify some other dampers to fit the nema 14s on the X axis 
On this topic i would like to ask if anyone thinks that the DRV8825 drivers are making the motors more noisy. I used DRV8825 on another printer and made the same high pitch noise along with high vibration compared to A4988. any thoughts?

pgilfernandez, 2016-02-29:
I have done the fixes suggested by Zorgonaute84 and they have really
reduced the noise on my Sigma.
Thanks!

Pablo Gil Fernández
email:
Twitter:
pgilfernandez@gmail.com
@exkr

Zorgonaute84, 2016-02-29:
Happy to Heard that.

I Will have a look in the Astrosyn Dumper too, thanks to share your tips.

BCN3D, when you will open a real forum (in english) to share this kind of tips and tricks ? I dont think that Github is the right place for that, but it's the only place to share between Sigma owners (and the S3D forum too)

kitusmark, 2016-02-29:
A lot of cool ideas arround here! The astrosyn dampeners are the key to solve the problem but a little bit expensive. We'll buy some and try them out. 

I found a replica on [thingiverse ](http://www.thingiverse.com/thing:664281)! 

About the forum... I don't think it's going to happen. We use to have one and didn't work out very well. Instead what we're looking is some people start communities @ [Google+](http://www.thingiverse.com/thing:664281) sharing their experiences. I think it's a good way to explain your work and development. 

What do you think?

mrcg, 2016-02-29:
All bigger 3d printer manufacture have their own. Even for my first K8200 the vendor has managed to have own forum. It was very useful to exchange findings and recommendations. Nowadays it should be not a big thing to setup own forum. All this useful informations should be shared between the users. Github in not the right forum, but unfortunately the only available.

pgilfernandez, 2016-02-29:
I do agree with you guys.
Marc, BCN3D Technologies should think again about a new forum site, maybe
the problem wasn't the low user activity but the bad user experience the
old forum had (I remember it and was hard to use...).
Please, give it a new opportunity.
=)

Pablo Gil Fernández
email:
Twitter:
pgilfernandez@gmail.com
@exkr

kapetanikolas, 2016-02-29:
I will have to agree with you guys. a forum is essential 
by the way i am selling astrosyn dampers and for sigma users ill send them at cost £5 if you get them from my site(https://emvioeng.com/shop/product-category/3d-printing/) just send a msg saying you have a sigma. Lest create a a kickass community to support each other  as the machine really has the potential to become SUPER!

Zorgonaute84, 2016-02-29:
Sure a forum is a must have. Maybe the old one was not a success because they wasn't Sigma section and the forum was mainly in Spanish !

rogeruceda, 2016-03-01:
Hi!

I am Roger from BCN3D, we will have a forum in our new website probably in 2-3 weeks. We are still fixing some things in the web. BTW do you like it?

Thanks for all this improvements. I am surprised in how fast people is improving the Sigma!

Regards

kapetanikolas, 2016-03-02:
hi guys
I like the new site a lot! and I am glad to read that you are planning a forum! 
I made a video of the noise reduction when a damper is added on the Y axis nema 17. https://www.youtube.com/watch?v=2gO24uwA1qU the first part is with the damper and the second without. keep you mouse on 5:50 and then press press 4 on you keyboard which should take you at 2:26. so you can go back and forth on the video to compare the noise.
I am planning to adapt the nema 17 dampers to fit the X nema14 motors. as soon as i have more ill report back

Zorgonaute84, 2016-03-02:
Yeah, really good new site ! But you've got some display and scrolling troubles.

rogeruceda, 2016-03-02:
Yes! We are still working on some issues

4truemodels, 2016-03-05:
Thx a lot Zorgonaute84 for sharing those tricks!
I will make my move next week in order to reduce noise level.
I'm not sure of have understood the use of dumpers kapetanikolas but I'll see further on that too.

I really like the new website design! Good work!

rogeruceda, 2016-03-05:
We will test the dumpers next week!

kapetanikolas, 2016-03-05:
Chris Lee has started a Sigma community for anyone who wants to join. https://plus.google.com/communities/100884192172984515424

ill be working on nema 14 vibration dampers this weekend in Bristol and try them on Monday.
The dampers on the Y axis reduced the high pitch noise and created a smoother sound.

Also tried my pop up enclosure last night...

[image]

rogeruceda, 2016-03-05:
OMG this is porn! :p

rogeruceda, 2016-03-11:
Hi guys,

We already have the forum in our website. Is still under construction buy you can start using it and giving us feedback.

Thanks!

https://www.bcn3dtechnologies.com/en/forums/forum/hardware/bcn3d-sigma/

AlejandroGarcia92, 2017-02-14:
We improved our stepper drivers to make less noisy the machine.

More info: https://www.bcn3dtechnologies.com/en/catalog/stepper-drivers-upgrade-kit/

closed 2017-02-14 by AlejandroGarcia92

pgilfernandez, 2017-02-14:
Hi Alejandro! Could you evaluate the noise reduction as a percentage? 50%, 60%...?

Thank you