BCN3D/BCN3DSigma-Firmware #51: Deactivation of motor and fan of inactive nozzle during single nozzle prints

kapetanikolas, 2016-02-29:
would it be possible to switch off the cooling fan and nema 14 stepper motor of the inactive nozzle during single nozzle prints?  

During single nozzle printing the fan and motor are not required to be active. maybe this could be done by setting a minimum temperature threshold (say 50 degrees) for the cooling fan and motor  to be activated.

This will make the printer quieter when switched on and not printing but also while printing with a single nozzle.

mrcg, 2016-03-01:
I have already raised this request to BCN3D support. This was my first idea to reduce the noise.
The board does not allow to control the speed from the extruder fan (only on/off). Fan speed control only for the noozle fan possible.

Another idea would be to add ntp fan controller board. Unfortunately it's difficult to find this for 24V fans which are used in the Sigma. For 12V fans some very small controller are available. At low temp (e.g. <50° fan off or only very low speed).

This should be changed in Sigma v2 for sure!!!!

kitusmark, 2016-03-10:
Hi @kapetanikolas. As @mrcg said, it is not possible to control the extruder fans as they're directly connected to the power supply. 

The unused NEMA14 of the X axis I think it can be unblocked and thus being cooler. We will try to implement that and do some test and see if on longer prints it doesn't move at all. 

Thanks.

AlejandroGarcia92, 2017-02-14:
We developed a upgrade kit according to your suggestion.

More info: https://www.bcn3dtechnologies.com/en/3d-printer/smart-fan-upgrade-kit/

closed 2017-02-14 by AlejandroGarcia92