BCN3D/BCN3DSigma-Firmware #52: Update FW Instructions in "Readme.md"

mrcg, 2016-03-04:
You should update the FW update instructions in the initial Readme.md to make it more clear which zip-file to be used for manual update.
Unfortunately i did not know at the beginning that i will find the compilied hex file in the "Source Code.zip" which you will only find when you go to "Releases".

I have downloaded the file from the repository (via RAW Button) and this file can not be used.
Make it more clear to the users who are not familar with Github.
Anyhow .... the latest FW Files and instructions must be placed on BCN3D Support pages.
At the moment only very old files are stored here.

kitusmark, 2016-03-07:
Hi @mrcg. I'll be updating the README file really soon, linking to the wiki properly. 
check this page and get back to me if something is not well explained. 

What file is the one you think is corrupted? the `.hex` file or someone from the LCD?
We're working on the web right now to include all the necessary info. It will be improving along the week.
Thanks a lot.

mrcg, 2016-03-07:
It would be useful to distribute the following informations:
1. Download from here: https://github.com/BCN3D/BCN3DSigma-Firmware/releases
2. Download the "Source Code.zip" package which contains required compilied FW-hex file and the LCD files. (Nobody expects to get compilied version in the source code package).

mrcg, 2016-03-08:
Well done. This will make like much easier for new users.

closed 2016-03-08 by mrcg