BCN3D/BCN3DSigma-Firmware #53: Measuring Vref on stepper motor driver boards?

kapetanikolas, 2016-03-04:
Hi, 
I touched the X motors and they are very hot so i just tweaked the on board potensiometers of the stepper driver circuits but want to get an accurate measurement of the current setting. Is there a Vref and if so how can i measure it?

kitusmark, 2016-03-16:
Hi @kapetanikolas. 

If you disassemble the stepper drivers, be careful with the flat flexible cable. And be careful with shorts because to measure you need to power the printer.

Answering to your question, yes there's a potentiometer to set the maximum current of the coils. You can measure the Vref voltage between the output of the pot and GND. The Current/Voltage ratio is 2. So if you set the pot to 0,5V you'll get 1 Amp of current. 

The output is the top pin and GND is the bottom left pin

[image]

Hope it helps. 

We're planning to release some kind of advanced mini-tutorial to tackle these kind of issues.

Zorgonaute84, 2016-03-22:
Is it a new version of the BCN3D Stepper driver on your picture ? Because mines are green and the setting screws are not in plastic...

kitusmark, 2016-03-22:
The driver is the same. The only changes are the silkscreen and the potentiometer (same value). The process and the results should be the same.

closed 2017-02-14 by AlejandroGarcia92