BCN3D/BCN3DSigma-Firmware #54: Purge settings

4truemodels, 2016-03-05:
Hi,
I have a suggestion about the "Purge Filament" settings...
Would it be possible to change the firmware in order to make the extruders move to the very same place as for "Remove Filament" when you press "Purge"?
By now I'm moving myself the extruders to that place so the filament would move a lot easier through the bowden tube during the purge process. I know it is not a big deal, but maybe it is not a bad idea to set up this. What do you think?

kitusmark, 2016-03-07:
Do you mean to the center - front position?

4truemodels, 2016-03-09:
Center-Front is not the position for "Insert Filament"?
I meant just Top-Front, where both extruders could be more "front" as would touch the metal box.
Anyway both positions would be much better than the one it has for the "Purge" function.
Thx!

guillemavila, 2017-01-21:
Hello, we are not implementing it because it would imply an extra home and therefore inefficient moves. 
As it's an interesting feature we leave the X/Y motors unblocked so everyone can move the heads to the position they want to. This way we don't add extra moves to the people who don't want to purge material at this position.

closed 2017-01-21 by guillemavila