BCN3D/BCN3DSigma-Firmware #55: [Issue] Purge Function in FW.1.4

Zorgonaute84, 2016-03-05:
I just updated my Sigma with the last version of the Firmware 1.1.4 and the Purge Function button is not working !!!

I havn't any troubles to insert or remove filament, but the Purge function is not available. The icon is here, but I can't touch on it... It's a very very big.

Thanks

kitusmark, 2016-03-06:
Hi @Zorgonaute84. Try to copy again the SD Files and it will be solved.

closed 2016-03-06 by kitusmark

sylus28, 2016-03-06:
Same issue here, try to copy again, reformat to FAT16 and copyn again : issue was not fixed

sylus28, 2016-03-06:
try to complete format and it works

Zorgonaute84, 2016-03-06:
@kitusmark, thanks to do the thing in the right order.
First someone create a issue report
Then someone provide a solution
Then the person who created the report say that his problem is solve
And finally the issue is closed...

My trouble is not fix after copy the file again, and the format (FAT16) don't solve my issue...