BCN3D/BCN3DSigma-Firmware #56: [Enhancement] Head position after a print

Zorgonaute84, 2016-03-05:
It could be better to place the head at the top front of the printer instead of at the back after a finished print (and a stopped print). Because at the back, their is some torsion on the bowden tube and the flat cable.

kitusmark, 2016-03-07:
Hi @Zorgonaute84 
You can achieve that by modifiying the end.gcode text in Cura software as follows:

```
;End GCode
M104 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)
G91                                    ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} ;move Z up a bit and retract filament even more
G28 X0                              ;move X/Y to min endstops, so the head is out of the way
G1 Y290                   ;move Y Axis to the front
M84                         ;steppers off
G90                         ;absolute positioning
;{profile_string}
```

That way everytime a print ends, the Y axis will go to the front of the machine.

Hope it helps.

Zorgonaute84, 2016-03-07:
Ah great idea, I didn't think about that ! The only things remain is we can't do that when we press stop during a print.

But I will implement that for end.gcode. thanks for the tips ;-)

closed 2017-02-14 by AlejandroGarcia92