BCN3D/BCN3DSigma-Firmware #57: [Issue]Extruder temperatures locked

sylus28, 2016-03-05:
with last firmware T0 is off and temperature stay @209°C and T1 is off stay @77°C
No issue for bed.
very strange issue,have printer form 2 days not understand all ,meanwhile I have an ultimaker 2 from 1 year and never have heating when command of regulation set to 0.
Thanks for assistance

kitusmark, 2016-03-07:
Hi @sylus28 

Have you managed to print something?
Please contact [technical support](mailto:sat@bcn3dtechnologies.com) and let me know when the problem is solved.

Thanks.

sylus28, 2016-03-07:
Yes on extruder 1 if the print temp is over 210,
Issue happens after few print folowing the firmware upgrade
Thermistor and connections checked .
I have identify bad ADC value on both extruder (93 and 770 respectivly)
I have contact my local reseller (france) we are on it I 'll let you inform.

kitusmark, 2016-03-07:
Ok. If it's firmware related it will be really really strange. We haven't change anything related with the temperature reading code. 

Sometimes it's a bad connection between the board in the carriage and the mainboard through the flat flexible cable. Remember if you mess with cables to turn off the printer first!

sylus28, 2016-03-07:
flat cables and probes was removed and reinsert printer off

sylus28, 2016-03-08:
EEPROM reset done according to reseler advice :no change issue still present

kitusmark, 2016-03-09:
Hi @sylus28. I've been talking to tech support. This issue happened when updating the firmware or it happened before? Just for trying, have you tried to load v1.0.0 again?

sylus28, 2016-03-09:
the issue appears few time after firmware update ( not suddenly)
trying to revert to each released firmware without success

sylus28, 2016-03-10:
machine return for repair,not so happy....

closed 2016-03-10 by sylus28