BCN3D/BCN3DSigma-Firmware #58: SD-Files v114

skroob, 2016-03-06:
when i update my sigma with Firmware 114 and i use the SD-files 114 then i can't use the purge-menu on the controlpanel from my sigma. i use now the sd-files from v113_d then i can use this menupoint.

sylus28, 2016-03-06:
had the same issue, do a complete format (FAT16) your SD and copy again 114 files and it woks

Zorgonaute84, 2016-03-06:
I tried to recopy the files without success,
then I tried to format (FAT16) my SD then recopy the fils, and it's worst !
Now I've got a message "Not Mounted drive"...

Zorgonaute84, 2016-03-06:
I reformated the SD Card from a Windows computer (instead of the last time from a MaC OS) I was able to fix the mounted trouble, but the Purge function still not aivalable...

Zorgonaute84, 2016-03-06:
Ok now it's works...
I reformat the card 3 times, and finally I used the SD.Filesv114.zip provide in the "release" section instead of the global download zip. Maybe there is a corrupted file somewhere...

skroob, 2016-03-06:
Ok thanks Zorgonaute... i use same the Files : (https://github.com/BCN3D/BCN3DSigma-Firmware/releases) and now it work for me too

kitusmark, 2016-03-07:
Hi @Zorgonaute84 @skroob and @sylus28. 

If now it's working for everyone i'm going to close the issue. If problems persist due to this issue, please reopen it and we will take a look at it. 

For the moment I'm going to update the documentation of the [wiki ](https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Updating-the-SD-Files-from-the-LCD-Display). 

Please take note of the warning about formatting SD Cards in Mac OSX. 

Thanks.

closed 2016-03-07 by kitusmark