BCN3D/BCN3DSigma-Firmware #59: Feature Request: Parallel print of same GCode on both extruder. 2x pieces with one print.

mrcg, 2016-03-08:
It would be really excellent feature to print same 3d model on both extruder in parallel.
In theory it should be possible but not sure if Sigma FW does support this.

The printing volume needs to be halved from 210x297mm to 105x297.
The benefit of this "duplication" feature is to print 2 pieces in one print.
This will save lot of time and will be unique feature on the market !!!!

pgilfernandez, 2016-03-09:
Hahaha, you have probably read my mind! since last week I had the same idea watching STACKER 3D printer on action, check this video:
[image]

It would be great if our printer can have this cool feature =)

4truemodels, 2016-03-09:
+1... or better... +1million!

kapetanikolas, 2016-03-10:
that will be a kickass feature!  it will make the machine very competitive

Zorgonaute84, 2016-03-22:
I don't think it's possible with the Sigma electronics. The Sigma is based on Arduibo, and Arduino is a single thread board. So it's not possible to provide the data for the boths extruders at the same time...

kapetanikolas, 2016-03-22:
So the 8 bit atmega is not man enough to drive at the same time 6 motor?  
Not even by fooling the controller? say by sending the same signals to both extruders-X motors and hot ends?

mrcg, 2016-03-29:
As a first step both extruders must move to it's initial position (Extruder_Left_X=50mm / Extruder_Right_X=150mm). Then the "feature" need to mirror all the X-axis stepper signals to both extruder (incl. filament extruder). Not sure if output can be mirrored in sync from the Sigma board.

If this in not possible then it would be possible (for sure) to do this "mechanically" by switch or relais.

The bigger challange i see in the correction of Noozle distance for both extruders in case they are differing more than 0,1mm.

kapetanikolas, 2016-04-29:
Ive been getting my self familiar with the firmware latelly and got to the parts of the dual X axis
there are motion modes that one can set I guess Mode 2 would do what we are discussing. For the Z offset of the nozzles I wonder if the higher nozzle could be told to make a raft first to compensate? 

[image]

closed 2016-09-30 by AlejandroGarcia92