BCN3D/BCN3DSigma-Firmware #60: Bug: "Purge Settings" cannot be entered after v01-1.1.4 upgrade

mrcg, 2016-03-08:
After upgrade to v01-1.1.4 (FW+SD) it's not possible to enter the "Purge Settings" (Utilities) menu any more. When i am pushing on the purge settings icon nothing is happening. Every other menu option is working.

sylus28, 2016-03-09:
https://github.com/BCN3D/BCN3DSigma-Firmware/issues/58

mrcg, 2016-03-09:
Which windows tool to be used to format propperly with FAT16?

sylus28, 2016-03-09:
Standar format tools (FAT) but unchecked quick format

kitusmark, 2016-03-09:
This works pretty well too. https://www.sdcard.org/downloads/formatter_4/

mrcg, 2016-03-09:
The root cause is not the FAT16 partition!!!!

The reason is that the SD.files are corrupt from the "BCN3DSigma-Firmware-01-1.1.4.zip" package.
The SD.files from the "SD.files.v114.zip" package are ok.

This is clearly visible if the checksum of the files from both packages will be compared ... non is matching !!!!!!

I do not understand why nobody was checking and correcting this base on the information provided from Zorgonaute84 few days ago. I have doubt that also the FW .hex value from the "BCN3DSigma-Firmware-01-1.1.4.zip" is corrupt as i have the issue to loose regulary all my EEPROM settings after v114 update.

**BEFORE RELEASING ANY NEW SW OFFICIALLY EXACTLY THIS PACKAGE'S NEEDS TO BE TESTED IN THE FACTORY !!!!!**

I am struggling since days after v114 update and the reason are corrupt SW release package!!!

[image]

kitusmark, 2016-03-10:
Hi @mrcg @sylus28 

I've changed the files and tested them. Please try to upload the firmware again through Cura-BCN3D (Machine > Install Custom Firmware). 

@mrcg we test everything. It has been a git problem as we are learning as we go. Thanks for all the detailed info.

mrcg, 2016-07-04:
Online package was corrupted. Issue fixed from BCN3D team.

closed 2016-07-04 by mrcg