BCN3D/BCN3DSigma-Firmware #62: XY calibration is useless

Whoopsss, 2016-03-10:
Hello,
On my Sigma I can properly calibrate by following the procedure, the X and Y calibration lines matches perfectly.
But the XY offset values doesn't seem to be stored by the printer, when I print I always have a bad offset in X and Y.
I tried to reload factory settings M502 + M500 and M500 after caibration, it changes nothing.
I have the problem in V113 and V114.

Thanks.

mrcg, 2016-03-10:
I have not exactly the same but similar issue. My Sigma is loosing all stored variables from EEPROM when switched off for longer time (e.g.15-20h).
You should check with M503 before and after calibration if values are changing and stored after power on/off.

mrcg, 2016-03-10:
BTW .... be aware that SD files from Source Package are corrupt. See my comments for issue #60.

kitusmark, 2016-03-10:
Hi @mrcg and @Whoopsss. I've changed the files. Please try to update again all the LCD SD Files and upload the .hex file again in Cura-BCN3D as refered in #60.

@Whoopsss Please connect the printer to pronterface and copy the results of the M503 command. 

Thanks

Whoopsss, 2016-03-10:
Hello Marc,

This is a screenshot of M503 gcode command.

[cid:image001.png@01D17AD5.4EADF0E0]

Thanks !

Best regards.

Jean François SEMA

Whoopsss, 2016-03-10:
In text format ;-)

Whoopsss, 2016-03-10:
In text format ;-)

>>> M503
SENDING:M503
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z1600.00 E152.00
echo:Maximum feedrates (mm/s):
echo:  M203 X250.00 Y180.00 Z15.00 E60.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X2250 Y2250 Z80 E800
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1500.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X5.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo: M301 LEFT P15.16 I1.16 D49.38 - M301 RIGHT P15.16 I1.16 D49.38
echo:Offsets (mm):
echo: X 303.70 Y 1.05 Z 0.15 Z probe-1.30
echo:Temp (ï¿½C):
echo: L_INSERT 220.00 L_REMOVE 170.00 L_BED 65.00 L_PRINT 215.00
echo:Temp (ï¿½C):
echo: R_INSERT 220.00 R_REMOVE 170.00 R_BED 65.00 R_PRINT 215.00
echo:LOG:
echo: prints 83.00 hours print 82.00 prints finished 18.00 max temp L 280.00 max temp R 280.00 max temp B 100.00

Whoopsss, 2016-03-10:
Hi,

After a new update of SD files and and .hex file, the preheat menu work like a charm ;-)

Thanks.

mrcg, 2016-03-10:
What you can try:
1. Delete the files "RepRap_T.d0F" and "RepRap_T.g0F" if those files are on SD. Restart printer and redo full calibration. Solved?
2. Format SD with FAT16 on windows PC (uncheck quick format). Then copy SD.files on SD. Download new package as the first one were corrupted. Solved?
3. Flash v114 FW manually based on description. Solved?

This are my ideas. Hope you will fix it soon.
Good luck.

closed 2016-06-01 by Whoopsss