BCN3D/BCN3DSigma-Firmware #63: Preheating menu no longer works

Whoopsss, 2016-03-10:
Hi, another strange issue.
This work early, then I got it was not working for custom materials but always for ABS and PLA, and now it does not work at all.
When I preheat, nothing happens.
I had this problem in V113, but the V114 upgrade changed nothing.

No problem for print.

Thanks.

mrcg, 2016-03-10:
You only have the issue (pre heating not working) when you select custom filament temperatures. When you use PLA or ABS you have no issue. Is my understanding correct???

I also use custom filament temperatures and have another very strange behaviour. Issue could be related to the custom temp setting commands used from the SD towards EEPROM.
But this only a suspect for the moment.

Whoopsss, 2016-03-10:
Hi,

After a new update of SD files and and .hex file, the preheat menu work like a charm ;-)

Thanks.

closed 2016-06-01 by Whoopsss