BCN3D/BCN3DSigma-Firmware #68: [Enhancement] adding recover print option

sylus28, 2016-03-28:
As Marlin Thinkergnome firmware https://github.com/TinkerGnome/Ultimaker2Marlin/releases ,it will be nice to implement recover print fonction with scan code file and restart the printing @ failed Z height

Wim1975, 2016-06-04:
+1
True, but then it would also be nice (necessary) if the Z height is shown during a print... so one knows where to restart it.

sylus28, 2016-10-20:
+1
And also an filament detector based on that https://www.youtube.com/watch?v=7Bhbh1mcsi8
(currently repetier)

AlejandroGarcia92, 2017-02-14:
We implement it for the upcoming new firmware version v01-1.2.2. 

Now avaliable in the Release Candidate. https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.2RC

closed 2017-02-14 by AlejandroGarcia92