BCN3D/BCN3DSigma-Firmware #70: MAJOR: M109/M190 Wait until temperature is reached does not work (1.1.4g=>1.1.5RC)

mrcg, 2016-04-25:
When sending M109 (Set Extruder Temperature and Wait) or M190 (Wait for bed temperature to reach target temp) the printer is proceeding with gcode without waiting for temperature to be reached.

This is causing the print code to be executed before bed and extruder is reaching it's target temperature.

I can not believe that this basic functionality is not working but i have tested this now several times.
Issue was happening with all 1.1.5 release candidates (1.1.4 not tested).

As workaround extruder and bed needs to be pre-heated before starting the print.

mrcg, 2016-04-30:
Just to make it more clear. The issue with M109/M190 only happens when printing via USB (e.g. Octoprint). When priting same code from SD the issue is not present.

Anyhow ... this is MAJOR issue when printing via USB (Octoprint) and needs to be solved urgently.

payala, 2016-05-18:
I can confirm this issue, it is really incredible that something like this is not working. It seems like nobody is printing through USB.

The same gcode that works nicely through the SD card fails to wait until heated through USB. Tried with pronterface, simplify3D and raw terminal and it just gives back the confirmation that the extruder/bed is heated even though it isn't yet.

ChrisTerBeke, 2016-05-23:
We're using the BCN sigma via [The Element](http://printr.com/the_element) (USB dongle) and also have this bug. Everything else is working perfectly :)

mrcg, 2016-05-29:
This bug is really annoying and limiting the possibility to use Octoprint with the Sigma.
I hope the issue will be fixed with highest priority.

I suspect the bug is related to the feature being able to modify the temperatures while priting from SD card. One good solution to overcome the issue when printing from USB could be to "pipe" the gcode commands coming from USB towards the SD reading interface. This would allow to control also tempereature and pause/stop via LCD when printing from USB.

kitusmark, 2016-06-08:
Hi @mrcg, @payala and @ChrisTerBeke,

Sorry for taking so long to address this issue but we're quite busy with other tasks. 
@AlejandroGarcia92 has fixed this issue and you can test it downloading the 1.1.6RC firmware version.

https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.1.6RC

If you don't know or remember how to upload the firmware manually, check[ this little guide!](https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Manually-changing-the-firmware-in-Cura)

Other things that have changed in this new version can be found [here](https://github.com/BCN3D/BCN3DSigma-Firmware/wiki/1.1.6RC)

Hope it helps!

mrcg, 2016-06-08:
First single extruder print via external USB was successfully. Now printer is waiting for bed and extruder to reach target temperature.

Not tested was dual extruder print.
Not tested was if there is any negative impact when printing from SD.

Thanks for the bugfix which will make the usage of Octoprint much more comfortable.

mrcg, 2016-07-04:
Issue solved with v1.1.6

closed 2016-07-04 by mrcg