BCN3D/BCN3DSigma-Firmware #71: Full calibration Impossible : Machine wants bed to reach 257°C?

Valenthior, 2016-05-14:
I wanted to do a full Calibration.
When the process wants to print the first Z pattern, the pinter stays stuck at "Adjusting temperatures, please wait".
As it was taking way too much time compared to previous "adjusting temperatures" I turned off the printer.
I Tried Multiple times to do the calibration procedure with the same issue.

Then I went to the "utilities/Temp/Preheat" Menu where I can see that the "preheat settings" are 257°C for bed and R Extr... Of course, As I am using PLA & PVA this set up is not mine.

My Z calibration seems lost and I can't do any print anymore...

mrcg, 2016-05-17:
What is the output from M503?

Valenthior, 2016-05-19:
[42](http://www.urbandictionary.com/define.php?term=42) ?
More seriously : what is M503?

Valenthior, 2016-05-21:
Problem seems to have been solved by a "reset to default value" through the use of:
http://koti.kapsi.fi/~kliment/printrun/
Using command "M502".

It would be interesting to have this kind of action within the touchscreen software instead of having to DL a specific software and plug a PC.

Kevinul, 2016-09-25:
I am having the same problem on firmware version 01-1.1.6.  
The calibration process gets stuck trying to heat the 2nd extruder and never finishes. Turning the printer off seems to be the only way out of the calibration process.

I tried reloading the firmware and that did not make a difference. 
I also tried installing the new release candidate version 1.2.0RC but cura would get stuck on the first step of "reading firmware.."  and never continues.  Tried in both cura version 0.1.4 and 0.1.5-beta2. Same result.
Is there a way to install new firmware directly with arduino software or another program?

Has anyone had this problem and found a solution?

This is the output I get from M503 in printrun
`>>> M503
SENDING:M503
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z1600.00 E152.00
echo:Maximum feedrates (mm/s):
echo:  M203 X250.00 Y180.00 Z15.00 E60.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X2250 Y2250 Z80 E800
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1500.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X5.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo: M301 LEFT P15.16 I1.16 D49.38 - M301 RIGHT P15.16 I1.16 D49.38
echo:Offsets (mm):
echo: X 304.00 Y -0.15 Z -2.90 Z probe-2.50
echo:Temp (ï¿½C):
echo: L_INSERT 220.00 L_REMOVE 170.00 L_BED 65.00 L_PRINT 215.00
echo:Temp (ï¿½C):
echo: R_INSERT 220.00 R_REMOVE 170.00 R_BED 65.00 R_PRINT 215.00
echo:LOG:
echo: prints 5.00 hours print 0.00 prints finished 0.00 max temp L 258.00 max temp R 275.00 max temp B 62.00`

Valenthior, 2017-02-05:
Problem solved using Temperature PID autotuning : M307 S230 E1
Where S230 is the temperature to do the AutoPID process and E1 is the extruder you want to tune. E1 = right extruder.

closed 2017-02-05 by Valenthior