BCN3D/BCN3DSigma-Firmware #72: Machine Control Panel in Simplify3D Connection lost...

Wim1975, 2016-05-19:
When I try to connect to the printer in Simplify3D... 
it first connects, 
suddenly there's a CRC error, 
then some commands that seem to work,

But then the printer suddenly reboots and... the connection is lost !

Here is wat the "Machine control panel" says : 

```
Attempting connection at \\.\COM4...
RECEIVED: 01-1.1.4
RECEIVED: echo:start
RECEIVED: BCN3D Sigma
  Testing plaintext communication protocol...
  Testing binary communication protocol...
  Testing alternate communication protocols...
RECEIVED: echo:Stored settings retrieved
  Attempting RTS reset and trying again...
RECEIVED: ok
  Connected to machine!
SENT: T0
WARNING: CRC mismatch detected
RECEIVED: 
RECEIVED: echo:Active Extruder: 0
Active Extruder: 0
RECEIVED: ok
SENT: M105
RECEIVED: ok T:17.7 /0.0 B:18.6 /0.0 T0:17.7 /0.0 T1:19.0 /0.0 @:0 B@:0
  Connection failed.
```

kitusmark, 2016-05-20:
Hi @Wim1975 !

Thanks for rising this issue. We've done this for future references:

[Connect BCN3D Sigma to Simplify](https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Connect-the-BCN3D-Sigma-to-Simplify-3D)

Please try and comment the issue and i'll close it if everything is ok

:+1:

Wim1975, 2016-05-20:
Solved, thank you BCN3D !
For others reading this : In S3D (Simplify3D) go to 
Tools>Firmware Configuration>Communication 
Then deselect "Allow command buffering".
This one was causing the trouble.

Wim1975, 2016-05-20:
Thanks, disabling the command buffering did the job. :-)

closed 2016-05-20 by Wim1975

kitusmark, 2016-05-20:
Awesome! :octocat:

HavocRC, 2022-03-16:
Enabling/disabling the buffer didn't work for me, however I found my issue!

I have a crappy Tronxy printer with Marlin on it. It connects just fine to other slicers but I was having issues with Simplify. Here's how I fixed it.

Tools -> Firmware Configuration -> Communication -> Uncheck "Wait for Startup Command". After I deselected this, my printer worked fine! I definitely has to do with the acknowledge command, but I'm too lazy to look for whatever it is and change it :).

Cheers!