BCN3D/BCN3DSigma-Firmware #74: Stuttering printing process causes blobs in version 1.1.15 !!!

Wim1975, 2016-06-01:
When doing some prints, the printer seems to pause while updating the LCD screen (temperature, process % and filename). For now I can somehow solve this by going to the "Settings" screen. The "Settings" screen doesn't need updating all the time and the stuttering stops and smooth prints come out, but temperatures and overall process cannot be followed then. It seems the updating from the LCD takes way too much processor time that is needed for a continuous printing process. Maybe leaving out the filename could solve this. (You should already know what you're printing since you chose it from the "Printing" screen, right?) I thought in previous versions the filename wasn't updated completely and I didn't see this behaviour. So I think quality is much more important than seeing the filename all the time.

mrcg, 2016-06-03:
Since the beginning i have realizied stuttering in some very data intensive gcode areas (e.g. very curvy model) when printing from SD card. I think the processing power get's on its limits sometimes.

This was another reason to use octoprint (RPi2) which has much more processing power. Unfortunately there is the bug with M109/M190 when printing from USB. 
So far no reaction on the bug report. :-(

kitusmark, 2016-06-08:
Hi @Wim1975 @mrcg 

We have found the same as @Wim1975. The LCD display takes so long to update the values and we've updated that part of the code to make it faster. You can test if it's fixed by downloading the 1.1.6RC firmware version.

https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.1.6RC

If you don't know or remember how to upload the firmware manually, check[ this little guide!](https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Manually-changing-the-firmware-in-Cura)

Other things that have changed in this new version can be found [here](https://github.com/BCN3D/BCN3DSigma-Firmware/wiki/1.1.6RC)

Hope it helps!

Wim1975, 2016-06-09:
It is indeed much better with the new version... (much faster). I close this issue.

closed 2016-06-09 by Wim1975