BCN3D/BCN3DSigma-Firmware #76: Add ON/OFF temperature control of hot-ends and bed in menus in stead of pushing 40 times + or - (enhancement)

Wim1975, 2016-06-09:
Hi, I wondered if it was possible to make the LCD temperature control "faster".
f.i. In the "Purge" menu... both hot-ends are being heated.
If I want to purge only one hot-end, it's pretty annoying I have to push the temperature [-] of the other hot-end about 40 TIMES !!! to turn it off... so... isn't it possible to add a function (ON/OFF) that puts on a hot-end to its default temperature (f.i. 215°C) or switch it off completely (0°C) ?

This could be implemented in all temperature based menus...

sylus28, 2016-06-09:
maybe a keyboard view when you press in the center of temp to set a custom value?

pgilfernandez, 2016-06-09:
I would suggest to make settings permanent, I mean, make the firmware remember your last selected temperatures. We usually work with similar temperatures over prints.
Anyway, I like the quick On/Off button for hotends and extruder, so +1 to this request

guillemavila, 2016-12-16:
Everything is working quite similar to all what you comment on version 1.2.1
Thanks for the feedback!

closed 2016-12-16 by guillemavila