BCN3D/BCN3DSigma-Firmware #77: M105 (Get Temperature) sometimes not reported correctly. Causing data gaps in octoprint.

mrcg, 2016-06-12:
When using Octoprint lot of data gaps are visible in the temperature graph.
Reason for this is that Sigma FW does not anser correct to M105 request.

########### Log 1 (causing data gap) #############

Send: N106205 M105*23
Recv: ok T:230.B@:47
Send: N106206 G1 X87.209 Y147.205 E1176.5076 F2280*11
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv:  T:0.1 /230.0 B:60.0 /60.0 T0:230.1 /230.0 T1:38.2 /0.0 @:71 B@:26
Recv: ok

########### Log 2 (causing data gap) #############
Recv: ok T:230.4 /230.0 B:60.1 /60.0 T0:230.4 /
Send: N84698 G1 X69.438 Y133.481 E961.5207*107
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv: 
Recv:  T:0. /0.0 B:./.T:0. /230.0 T1:38.3 /0.0 @:66 B@:21
Recv: ok

########### Log Good Case 1 #############
Send: N81171 M105*41
Recv: ok T:229.5 /230.0 B:59.9 /60.0 T0:229.5 /230.0 T1:38.2 /0.0 @:81 B@:39

########### Log Good Case 2 #############
Send: N89691 M105*40
Recv: ok T:230.1 /230.0 B:59.9 /60.0 T0:230.1 /230.0 T1:38.4 /0.0 @:71 B@:44

########### Log Good Case 3 #############
Send: N92746 M105*41
Recv: ok T:230.0 /230.0 B:60.0 /60.0 T0:230.0 /230.0 T1:38.3 /0.0 @:73 B@:24

Screenshot for data gaps in Octoprint:

[image]

mrcg, 2016-09-26:
Issue still present in final 1.2.0.
Anyhow it's only minor (cosmetic) issue.

mrcg, 2017-07-16:
Issue still open since more than one year.
Problem still exists with FW v1.2.4 and latest motherboard revision. 

[image]

AlejandroGarcia92, 2017-07-18:
Sorry, we are not focused on fix this issue by now.

mrcg, 2017-07-18:
Better to get honest answer than no answer ... even if i do not like this.
To fix the root cause might be quite difficult, i can imagine.

BUT

You could add plausibility check before sending the M105 information.
This should be quite easy to be detected.
If the data are not consistent than it should be discarded and not be send.
Better to have not M105 data rather than wrong/faulty one.
Discarding wrong M105 information would not have negative impact to statistics from external tool.

WDYT?

mrcg, 2017-07-18:
I was so much surprised that you mentioned it will NOT be fixed.
Originially i was thinking it is FW issue.

Your answer was making me thinking why you might not be able to fix this.
My suspect now is that it's HW issue (flat cable connection) because the issue is almost only visible for the moving tool. If this is the case than it looks like general flat cable connection issue. 

I really hope that broken thermistor will be detected for secutriy reason in case the flat cable connection is "broken" for longer (few seconds) time of period. Can you confirm this?

Anyhow, please check my statement from above which could minimize the impact in statistics.