BCN3D/BCN3DSigma-Firmware #78: Extra menus for CALIBRATION : (1) Z LEFT (2) Z RIGHT (3) X/Y BOTH please ! (time saving enhancement)

Wim1975, 2016-06-29:
Is it possible to add extra calibration menus ?
1) FULL CALIBRATION (=2+3+4+5)   OK
2) BED CALIBRATION                        OK
3) Z CALIBRATION LEFT                <<
4) Z CALIBRATION RIGHT             <<
5) X/Y CALIBRATION BOTH           << 
( or the last one 5) X CALIBRATION BOTH and 6) Y CALIBRATION BOTH)

This is helpful when 
- changing a nozzle (to f.i. 0.6 mm) for only ONE SINGLE EXTRUSION print.
(Bed and XY calibration not needed)
- the bed is already (manually) better adjusted then when using the automatic calibration (I have to do that all the time) so I want to skip the bed calibration.
- the full calibration is almost perfect and only a small adjustment in one of the parameters needs to be made

Please !

sylus28, 2016-06-30:
+1000

AlejandroGarcia92, 2016-06-30:
Now we are implementing a new screen interface that it includes a skip button for each Calibration step. In addition, we are going to include an extra menu where you will be able to adjust manually the calibration values.
Finally, once we finish implementing these features, we will release a RC version.

Thank you for the feedback.

sylus28, 2016-06-30:
Nice feature on my UM2 Z baby stepping is helpfull for the first layer specially when you change nozzle or bed

Wim1975, 2016-06-30:
Thank you Alejandro Garcia !

guillemavila, 2016-12-16:
All features implemented (v1.2.1)

closed 2016-12-16 by guillemavila