BCN3D/BCN3DSigma-Firmware #79: Unknown travel Issues

Lostinthe907, 2016-06-30:
[Video.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/340656/Video.zip)
[twoColor1_2_7_10_53new.txt](https://github.com/BCN3D/BCN3DSigma-Firmware/files/340657/twoColor1_2_7_10_53new.txt)
Hello, I'm am having an unknown Issue I am hoping someone can help me fix. 
 this started with my first model, So I updated firmware and tried again. still the same issue is happening. 
 While Printing In the Middle of a layer, it will travel to the left or to the front of the printer. leaving a small line of filament to extrude and attach back to the print. I have attached a video to help show what's going on, Because I cannot explain it well enough. 
 Not sure if is caused buy having to high of a Min layer time or some other issue. 

any help would be really grateful. 
 I am Running the PLA profile for IDEX std. 
Okay I have Uploaded the Gcode file for help. the issue happens 2 times in just printing the Skirt. 
 Maybe someone who understands the gCode better then I do can see the issue their. 

-Levi

mrcg, 2016-07-04:
Strange. The gcode looks pretty normal in gcode viewer (Simplify3D).

1. Which FW version do you have on the Sigma?
2. Are you printing from SD or external USB? If USB, which SW?
3. Does it happen always on exactly the same layer or randomly?
4. From where do you have the gcode? Did you slice on your own? Which Slicer?

In the gcode i can see that "G92 E0" (zeroing of extrusion distances).
I urgently recommend not to use this as i have phased some issues with this.

You should try to remove all lines with "G92 E0" and print again ... issue fixed?
Just use the search and replace function of any text editor to do this in 10 seconds.
Replace "G92 E0" with e.g. ";"

Good luck.

Lostinthe907, 2016-07-04:
I am using the latest firmware, printing from SD, and it happens randomly. Sometimes twice in a row. I am using Cura. 
The G92 E0, is there a setting in cura to change this? If not I will try removing and printing again. 
BCN3D seems to think it's my SD card reader. But I'm trying a new SD card first. Seems okay so far (5 minutes into print). If I end up removing the code and printing again I'll update what's going on.

mrcg, 2016-07-04:
Hope it's SD and you have it fixed now. ;-)

For Cura i do not know but in S3D there is an option to activate/deactivate the "G92 E0" usage.
It's just an idea but not sure if it is your issue.

Lostinthe907, 2016-07-04:
Me too. Think it may actually have been the Micro adapter. But I will try it. 

Thanks!!

guillemavila, 2016-12-16:
Solved

closed 2016-12-16 by guillemavila