BCN3D/BCN3DSigma-Firmware #84: Bug: Dual print not aligned when printing from USB.

mrcg, 2016-07-17:
When printing gcode with dual extrusion via USB (e.g. Octoprint) then T0 and T1 are not aligned according M530 calibration settings. When printing the same gcode via SD everything fully aligned according to calibration settings.

Issue verified with v1.1.6 and v1.2.0_RC.
[image]

mrcg, 2016-09-25:
Issue still not fixed in v1.2.0 (from 23.09.2016).
This bug makes dual printing via USB impossible.

AlejandroGarcia92, 2016-09-26:
Can you share with us this gcode file?

mrcg, 2016-09-26:
Attached the gcode file. But this happens also with any other gcode file (dual extruder).
When printing via SD everything is ok and aligned. Printing same gcode via USB (octoprint) the extruders are not aligned.

https://dl.dropboxusercontent.com/u/20305979/github/calib_cube_3x3_layer.gcode

AlejandroGarcia92, 2016-09-27:
We got the same issue but with more shift in Y axis than X, always by USB printing using this gcode. So we've tested some gcodes generated by Cura and we had no shift issues. Then we are going to still investigating what happens with it.

mrcg, 2016-09-27:
I have sliced the same object also with Cura (everything default setting) and i have the same issue when printing via USB. Does not seem to be Simplify3D specific issue. When printing same gcode via SD everything is ok.

GCode generated with CURA (also not aligned when printing via USB):
[image]

Cura GCode:
https://dl.dropboxusercontent.com/u/20305979/github/calib_cube_3x3_layer_cura.gcode

Could it be possible that FW does not take calibration data into consideration or using defaults from wrong motherboard (as defined in Configuration.h) when printing via USB?

My calibration data (M503):
Recv: echo: X 304.15 Y 0.05 Z -0.30 Z probe-1.70

AlejandroGarcia92, 2016-09-28:
I have got those results:

[image]

My calib data:

mrcg, 2016-10-06:
Any finding/progress on this issue?

mrcg, 2016-11-14:
What about correction for this issue?

AlejandroGarcia92, 2016-11-18:
Doing some test in other printers, we got successful results without this kind issues. For us is difficult to fix this issue if we can not reproduce it. Anyway, we are aware of that issue.

mrcg, 2016-11-18:
Beeing aware about this issue does not help me. I need a fix for this.
You already reproduced the issue and you should be able to analyze the root cause.
I need a solution/correction for this.

AlejandroGarcia92, 2016-11-23:
I think that the problem is caused by mechanical origins but i hope that we can fix it via firmware. I will tell you my thoughs about it and also this will be added in the current RC firmware version ASAP.

AlejandroGarcia92, 2016-11-24:
Firmware RC version updated. https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.2RC

mrcg, 2016-11-26:
Tested v1.2.2RC right now.
X-Axis seems to be solved now.
Unfortunately i have now shift in Y-Axis.
Can you fix this as well ???

[image]

mrcg, 2016-12-07:
I can not believe that it's so difficult to fix this issue.
This is really k.o. criteria to use commonly used printserver like octoprint.

AlejandroGarcia92, 2016-12-16:
It seems that the problem happens because the printer doesn't block axis during change tool by USB and some vibrations shift it. Soon(Next Week), I'm going to update our RC version and I hope that it will work properly.

AlejandroGarcia92, 2016-12-23:
@mrcg I have updated the version.

mrcg, 2016-12-23:
@AlejandroGarcia92 Thanks for the "christmas present".

I will test it next week. Many thanks for your support.
I really hope it's fixing the issue. ;-)

mrcg, 2017-01-06:
I have done some dual print via USB and the issue is fixed with latest 1.2.2RC.
**Great and many thanks for fixing this issue!!!**
The correction should be available in next official realease.

Issue to be closed now!!!

guillemavila, 2017-01-07:
Awesome!
Thank you for the feedback, that's our goal bring you the most value from the Sigma :)
Yes it will be available in the next official update.

Have a nice printing Saturday!

closed 2017-01-07 by guillemavila