BCN3D/BCN3DSigma-Firmware #86: Critical Bug in v1.2.0_RC: Print via USB is continuously failing

mrcg, 2016-07-22:
With v1.2.0_RC it's simply impossible to print detailed model (lot of gcode per seconds) via USB.
All the prints are failing sporadically with error:
# Error:Line Number is not Last Line Number+1

In the terminal log you can see that some lines are not getting acknowledged from the Sigma which is causing the print to fail. Issue with same gcode happens always on different lines which could be indication about performance/buffer issue.

It's really a pitty that basic functionality (printing via USB) has so much bugs!!!
Before optimizing the LCD GUI this issues should be fixed.

################### Log 1 ################
Send: N617 G1 X117.278 Y114.781 E3.9621*93
Recv: ok
Send: N618 G1 X118.037 Y115.510 E3.9774*94
Recv: ok
Send: N619 G1 X118.077 Y115.551 E3.9782*87
Recv: ok
Send: N620 G1 X118.333 Y115.841 E3.9838*92
Send: N621 G1 X118.370 Y115.886 E3.9847*89
Recv: ok
Recv: ok
Send: N622 G1 X118.557 Y116.137 E3.9892*81
Recv: ok
Send: N623 G1 X118.737 Y116.246 E3.9923*90
Send: N624 G1 X118.856 Y116.337 E3.9945*82
Recv: ok
Send: N625 G1 X119.008 Y116.481 E3.9975*88
Send: N626 G1 X119.206 Y116.588 E4.0008*82
Send: N627 G1 X119.412 Y116.757 E4.0046*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N620 G1 X118.333 Y115.841 E3.9838*92
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N621 G1 X118.370 Y115.886 E3.9847*89
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N622 G1 X118.557 Y116.137 E3.9892*81
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N623 G1 X118.737 Y116.246 E3.9923*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N624 G1 X118.856 Y116.337 E3.9945*82
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N620 G1 X118.333 Y115.841 E3.9838*92
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N621 G1 X118.370 Y115.886 E3.9847*89
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N622 G1 X118.557 Y116.137 E3.9892*81
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N623 G1 X118.737 Y116.246 E3.9923*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
Recv: ok
Send: N624 G1 X118.856 Y116.337 E3.9945*82
Recv: Error:Line Number is not Last Line Number+1, Last Line: 619
Recv: Resend: 620
[... 401 lines cut ...]
Send: N624 G1 X118.856 Y116.337 E3.9945*82
Recv: ok
Send: N625 G1 X119.008 Y116.481 E3.9975*88
Recv: ok
Send: N626 G1 X119.206 Y116.588 E4.0008*82
Recv: Error:checksum mismatch, Last Line: 620
Recv: Resend: 621
Recv: ok
Send: N621 G1 X118.370 Y115.886 E3.9847*89
Recv: Error:Line Number is not Last Line Number+1, Last Line: 620
Recv: Resend: 621
Recv: ok
Send: N622 G1 X118.557 Y116.137 E3.9892*81
Recv: Error:Line Number is not Last Line Number+1, Last Line: 620
Recv: Resend: 621
Recv: ok
Send: N623 G1 X118.737 Y116.246 E3.9923*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 620
Recv: Resend: 621
Recv: ok
Send: N624 G1 X118.856 Y116.337 E3.9945*82
Recv: Error:Line Number is not Last Line Number+1, Last Line: 620
Recv: Resend: 621
Recv: ok
Recv:

mrcg, 2016-09-17:
Will this issue be fixed in final 1.2.0 ???
This bug makes dual printing via USB (Octoprint) not possible.

AlejandroGarcia92, 2016-09-19:
No, but we are stil working on this issue.

ChrisTerBeke, 2016-09-20:
Confirmed here as well using The Element and 1.2.0-RC0. Why is this happening? Seeing that you based your firmware on Marlin, which works perfectly fine out of the box, I assume it's one of your changes? USB printing is still an important feature, so it would be great if this could be fixed soon after months of waiting.

AlejandroGarcia92, 2016-09-20:
We have not done the release yet so we are focused on it to fix this issue. If we don't get it at time, we will do another the firmware update ASAP.

ChrisTerBeke, 2016-09-21:
Any clues where the issue might be? I wanna take a look but don't know where to start reading the code.

AlejandroGarcia92, 2016-09-23:
We fixed this issue. Then, the new firmware version 1.2.0 is ready but the release is going to be next Monday. So please, Download the newest version from master or 1.2.0RC branch and send us feedback if this issue is still happening or not ASAP.

ChrisTerBeke, 2016-09-23:
Cool, will try that today!

ChrisTerBeke, 2016-09-23:
Okay, just did some testing. It does communicate over USB now, respects heating commands etc and sends all the lines without skipping them. But it stutters a lot, which makes me think the buffer on the printer side is not large enough. Used Repetier-host to test, and it seems that every line sent to the printer is executed immediately without buffering a few lines. There is a setting in Marlin to increase the buffer size (if that is indeed the problem).

AlejandroGarcia92, 2016-09-23:
The printer uses a 4 lines of buffer, it's the same buffer for sd printing. I have no problems with Repetier-host in my print test. I have configured Repetier-host just like that:
[image]

If you can share with us a gcode to test, it could be useful for us.

ChrisTerBeke, 2016-09-23:
Those settings seem to have improved the movements. But now I get this in my repetier console:

[image]

mrcg, 2016-09-25:
**I have tested the latest v1.2.0 (from 23.09.2016) and the issue seems to be fixed.    :-)**
About 8h of USB printing (via Ocroprint on Raspberry 2) wihout this issue. **Thanks for fixing this major issue.**

Anyhow two issues are still not fixed in 1.2.0:
MAJOR: #84 
Title: Bug: Dual print not aligned when printing from USB. ===> Please fix this major bug to make dual printing possible via USB. It was promissed that this issue will be solved in next (1.2.0) release. Unfortunately not.

MINOR: #77 
Title: M105 (Get Temperature) sometimes not reported correctly. Causing data gaps in octoprint. ===> Still data gaps in Ocroprint. It's only cosmetic issue (minor thing) but should be fixed as well.

mrcg, 2016-09-26:
Issue solved in final v1.2.0 firmware.

closed 2016-09-26 by mrcg