BCN3D/BCN3DSigma-Firmware #94: Remove filament at the end of g-code

omersa, 2016-08-30:
When done printing I leave black PLA filament in the printer. But after some time (few hours or a day) when I wish to print again a few seconds after the beginning of print, the PLA breaks just a few centimeters under it's driving motor. So if you don't hear the crack, some of model is printed and then there is no more filament.

My question is: is there any g-code command (besides retract xxx mm) to remove the filament from the hotend and the tubes completely after finishing the print? If not is there a way to add some new command in firmware?

If not of the above, what would be the right speed and length of retract to remove the fillament in whole? G1 Fxxx E-yyy

omersa, 2016-08-31:
Ending script:

M140 S0     		;heated bed heater off (if you have it)
G91       		;relative positioning
G1 Y+10 F12000	;move Z up a bit
G28 X0 Y0     		;move X/Y to min endstops so the head is out of the way
M109 S240		;heat extruder to 240 (PLA)
G92 E0        		;zero the extruded length
G1 F3000 E-1000	;remove filament
G1 F12000
M104 S0		;extruder off
M84         		;steppers off
G90

guillemavila, 2016-12-16:
We're not implementing that option in our default profiles as almost anyone of the filaments we use have this problem. Some of the early versions of Colorfila spools presented that issue but not the ones we are selling the last months.

Anyways if you need that feature the right way to implement it is by using the gcode you share. Thanks!

closed 2016-12-16 by guillemavila