BCN3D/BCN3DSigma-Firmware #98: Laptop not connecting to main board to update Firmware

ChaosRegent, 2016-09-20:
I cannot update the firmware to the main board because my Laptop (running Windows 7 Enterprise) will not connect. Getting an error in Device Management screen on the FTDI (COM7) port. When I attempt to upgrade the device driver it says i have the most up to date driver available.

m0w1337, 2016-10-04:
Hi ChaosRegent,
I don't know if this is the same issue i observed, but for me it worked as soon as i put a USB hub between my Laptop and the Sigma. The reason for these kind of issues might be the following little hardware bug on the mainboard:
D8 should be left unassembled (so can L2 and F3) and instead of connecting the RESET pin of IC2 to +5V this pin should be the only one connected to VBUS (via a 4k7 resistor) and to GND via 10k pull down to ensure the handshake gets initiated correctly as soon as the host connects to the Sigma ;) (maybe this will get into a new PCB revision when the current batch is used up)

Good luck with the Hub!

kitusmark, 2016-10-10:
Hi @ChaosRegent, 
It's strange because if your PC has assigned a COM port to the printer it seems to be connecting. Be sure to double check FTDI drivers stuff. We work with Win7 so I have no idea how to Win8 and up handles that.

kitusmark, 2016-10-10:
Hi @m0w1337,
Can you open an issue in the BCN3D-Electronics repo with what you just commented? 
That way we can discuss it without affecting the firmware repo! 
I'll be revisiting the design shortly and I'd like to take in consideration all the feedback. 

Thanks !

closed 2017-04-05 by AlejandroGarcia92