BCN3D/BCN3DSigma-Firmware #99: Version 1.2.0 issues

omersa, 2016-09-26:
This is just my opinion, but am asking all others, what do they think:
- Preheating setting is now much better, because you can choose extruders or heated bed with just one touch, BUT you can not set temperature manualy any more. It is a big problem for me, as I am printing big models and wish to have a temperature 50 degrees on heated bed. The setting is 65, which means, that there Will be temperature difference when I will print and this can affect adhesion.
- File manager is better looking, BUT before the files were sorted in creation order - newer files first. This was better, so you could select the last file on SD card very quickly. Now you might have to list through the whole SD, also because the folders are listed first.
- In manual for updating SD files for LCD display you should add something like that: If you updated SD files and after that the printer start-up procedure is very slow, you should format the SD card and copy SD files once again. Be sure to eject the card from computer.

haqbany, 2016-09-28:
Thank you for the update about the LCD display i was going crazy :-)

adrienciejak, 2016-09-29:
Hello BCN
This update look amazing!
But I've just try to update my 2 printers but it's not a succes.
When 2 printers start with new firmware the starting animation go slow and after the number of firmware version there is a fix. Nothing happen!
Can you help me please....?
Thanks

AlejandroGarcia92, 2016-09-29:
@adrienciejak,  I supose that the firmaware was uploaded to the mainboard successfully, so try the following:

Format the LCD SDcard, always is FAT16.

Finally copy again the SDfiles of the new update.

Send me feedback, regards.

adrienciejak, 2016-09-29:
Hello Alejandro
It works perfectly!
Thanks a lotAdrien

Lostinthe907, 2016-09-30:
I am Using windows 10 and am having the problem of it loading slow.  and not at all. I want to format the SD card as FAT16 But I'm not sure the Best way. Windows 10 gives me 4 options; NTFS, FAT, FAT32(default), And exFAT. I do not want to use the wrong one. I also have SDFormatter version4 Installed. Any help would be Awesome. would really like to get printing ASAP. :)

omersa, 2016-09-30:
FAT is FAT16

haqbany, 2016-09-30:
You need FAT16 i use it with windows 10 , you should format the SD card FAT16 and copy SD files once again. Be sure to **eject** the card from computer.

Lostinthe907, 2016-09-30:
@omersa Thanks. the FAT in windows 10 worked correctly. 
Happy to be printing again!

pvnct, 2016-10-01:
Print time is stuck to 1 min on every print. no effect on the print. 

It's not a big issue but is there somebody else having the same ?

AlejandroGarcia92, 2016-10-04:
@pvnct it could be that failed during the parsing to extract the print time. Send me the gcode and I will say you something for sure.

pvnct, 2016-10-04:
@AlejandroGarcia92 I think you are right, because i don't have the problem with gcodes from BCN3D's cura. As i am on mac i was using the regular version of cura for mac.

So problem solved

haqbany, 2016-10-04:
I have smiler problem the print god worng !

AlejandroGarcia92, 2016-10-04:
@haqbany  Can you explain what happens more in detail?

m0w1337, 2016-10-04:
The Time issue arises as soon as the print takes 1 Hour or longer, this is simply due to the selected CURA Language (in preferences).
As CURA translates the "xx hours xx minutes" within the gcode file to the selected language, but the sigma only understands english. So Set your CURA to english and everything is fine.

haqbany, 2016-10-04:
Here is whats happens when i print !
some time @AlejandroGarcia92 
[image]
[image]

haqbany, 2016-10-04:
@AlejandroGarcia92 i don't know why ? this the file http://www.thingiverse.com/thing:856101
best regards

AlejandroGarcia92, 2016-10-04:
@haqbany Now, I have no idea. Send me your gcode file that you generated because the .stl file is useless for me. I will send you feedback ASAP after test your gcode file.

omersa, 2016-10-05:
Dear friends, are you sure this is a kind of writing that is suitable for my post? I was just discussing new FW options. Please open your own issue with your kind of problems.

closed 2017-04-05 by AlejandroGarcia92