BCN3D/BCN3DSigma-Firmware #101: Suddenly "Cooling Down Heater" when printing through USB

karimcitoh, 2016-10-03:
I am printing a file through USB, and keeping track of the communication between the BCN3DSigma and my computer. Without doing anything in the printer itself, it suddenly turned heater off.

```
Sending command ----> N3559 G1 F1800 X120.903 Y70.168 E1134.46664*55
<---- Recv: ok
Sending command ----> N3560 M105*23
 <---- Recv: ok T:214.9 /215.0 B:60.0 /60.0 T0:25.3 /0.0 T1:214.9 /215.0 @:80 B@:44
Sending command ----> N3561 G0 F5400 X121.398 Y70.168*70
<---- Recv: ok
Sending command ----> N3562 G1 F1800 X176.346 Y125.116 E1136.21366*11
 <---- Recv: ok
Sending command ----> N3563 G0 F5400 X176.440 Y124.715*120
 <---- Recv: ok
Sending command ----> N3564 G1 F1800 X121.892 Y70.167 E1137.94796*62
 <---- Recv: ok
Sending command ----> N3565 G0 F5400 X122.387 Y70.167*64
<---- Recv: ok
Sending command ----> N3566 G1 F1800 X176.519 Y124.299 E1139.66903*3
 <---- Recv: ok
 Sending command ----> N3567 G0 F5400 X176.557 Y123.842*113
 <---- Recv: ok
Sending command ----> N3568 G1 F1800 X122.881 Y70.166 E1141.37561*48
<---- Recv: ok
 Sending command ----> N3569 G0 F5400 X123.376 Y70.166*66
<---- Recv: ok
 Sending command ----> N3570 G1 F1800 X176.595 Y123.385 E1143.06765*14
<---- Recv: ok
Sending command ----> N3571 G0 F5400 X176.633 Y122.929*122
<---- Recv: ok
 Sending command ----> N3572 G1 F1800 X123.871 Y70.166 E1144.74518*57
<---- Recv: ok
Sending command ----> N3573 G0 F5400 X124.365 Y70.166*76
<---- Recv: ok
 Sending command ----> N3574 G1 F1800 X176.671 Y122.471 E1146.40819*13
 <---- Recv: ok
Sending command ----> N3575 G0 F5400 X176.709 Y122.014*113
 <---- Recv: ok
 Sending command ----> N3576 G1 F1800 X124.860 Y70.166 E1148.05666*58
 <---- Recv: ok
 Sending command ----> N3577 G0 F5400 X125.355 Y70.166*74
 <---- Recv: ok
 Sending command ----> N3578 G1 F1800 X176.747 Y121.558 E1149.69062*12
 <---- Recv: Cooling Down Heater
 <---- Recv: ok
 Sending command ----> N3579 G0 F5400 X176.785 Y121.101*127
 <---- Recv: ok
 Sending command ----> N3580 G1 F1800 X125.850 Y70.165 E1151.31006*61
 <---- Recv: ok
```

So the printer keeps "printing" and performing the next commands, but it can't print:

```
Sending command ----> N3655 G0 F5400 X144.154 Y70.155*77
<---- Recv: echo: cold extrusion prevented
<---- Recv: ok
Sending command ----> N3656 G1 F1800 X176.787 Y102.788 E1200.17397*15
<---- Recv: ok
Sending command ----> N3657 G0 F5400 X176.787 Y102.293*123
<---- Recv: echo: cold extrusion prevented
<---- Recv: ok
Sending command ----> N3658 G1 F1800 X144.649 Y70.155 E1201.19577*54
<---- Recv: ok
Sending command ----> N3659 G0 F5400 X145.144 Y70.155*65
<---- Recv: echo: cold extrusion prevented
<---- Recv: ok
 Sending command ----> N3660 G1 F1800 X176.787 Y101.798 E1202.20183*9
<---- Recv: ok
Sending command ----> N3661 G0 F5400 X176.787 Y101.303*117
<---- Recv: echo: cold extrusion prevented
<---- Recv: ok
Sending command ----> N3662 G1 F1800 X145.638 Y70.154 E1203.19218*53
<---- Recv: ok
Sending command ----> N3663 M105*23
<---- Recv: echo: cold extrusion prevented
<---- Recv: echo: cold extrusion prevented
<---- Recv: ok T:102.7 /0.0 B:50.6 /0.0 T0:25.4 /0.0 T1:102.7 /0.0 @:0 B@:0
```

AlejandroGarcia92, 2016-10-04:
Sending command ----> N3578 G1 F1800 X176.747 Y121.558 E1149.69062*12
 <---- Recv: Cooling Down Heater
 <---- Recv: ok

According to these lines, it's activating a inactivity system witch shuts down the heaters. So, send me the gcode and I will say you what happens. My predictions:  I think that there are missing some important lines at the beginning of the gcode such as:

Some thing like that(by Cura):
M140 S50.000000
M109 S220.000000
M190 S50.000000
;Sliced at: Fri 03-06-2016 18:27:15
;Basic settings: Layer height: 0.2 Walls: 0.8 Fill: 15
;Print time: 11 hours 51 minutes
;Filament used: 19.834m 156.0g
;Filament cost: None
;M190 S50 ;Uncomment to add your own bed temperature line
;M109 S220 ;Uncomment to add your own temperature line
G21        		;metric values
G90        		;absolute positioning
M82        		;set extruder to absolute mode
M107      		;start with the fan off
G28 X0 Y0  		;move X/Y to min endstops
G28 Z0     		;move Z to min endstops
G92 E0        	;zero the extruded length
G1 Z5 F200		;Safety Z axis movement
G1 F50 E7     	;extrude 7mm of feed stock
G92 E0        	;zero the extruded length again
G1 F200 E-4		;Retract before printing
G1 F12000
M117 Printing...	;Put printing message on LCD screen

Or (by simplify) :
 
G90
M82
M106 S0
M140 S100
M190 S100
M104 S245 T0
M104 S230 T1
M109 S245 T0
M109 S230 T1
G21       	;metric values
G90       	;absolute positioning
M107       	;start with the fan off
G28 X0 Y0  	;move X/Y to min endstops
G28 Z0     	;move Z to min endstops
T1          	;Switch to the 2nd extruder
G92 E0      	;zero the extruded length
G1 F100 E10 	;extrude 10mm of feed stock
G92 E0      	;zero the extruded length again
G1 F200 E-9
T0         	;Switch to the first extruder
G92 E0      	;zero the extruded length
G1 F200 E10 	;extrude 10m of feed stock
G92 E0      	;zero the extruded length again
G1 Z5 F200	;Safety Z axis movement
G1 F[travel_speed]
G92 E0
G1 E-4.0000 F2400
G1 Z0.300 F1002
; layer 1, Z = 0.25
 T0			;start tool switch 0
 G1 Y250 F12000		;travel
 G1 F500 E-0.5			;fast purge 			
 G1 F50 E0.5			;slow purge
 G92 E0			;reset t0
 G1 F3000 E-4.5		;retract
 G1 F12000		;end tool switch
T0
; tool H0.250 W0.480
; skirt
G1 X76.848 Y117.262 F12000
G1 Z0.250 F1002
G1 E0.0000 F720
G92 E0

M109 heats the hotends.
M190 heats the bed.
G28 does home.

Those gcodes disable the inactivity system. Also it seems that you heat all using preheat, witch enable that inactivity system.

karimcitoh, 2016-11-14:
Hi Alejandro, thanks for your response, and excuse me for missing it until now.

Yes, M109 and M190 heats the hot ends. Are you sure that G28 also disables the inactivity system after homing?

I will perform some more tests this week and I will let you know!

AlejandroGarcia92, 2016-11-15:
Yes, also M109 and M190 disables the inactivity system.

karimcitoh, 2016-11-23:
Solved the issue by sending `G28` at the beginning, and adding `M85 S0` to my start gcode.

closed 2016-11-23 by karimcitoh