BCN3D/BCN3DSigma-Firmware #103: Potential bug in M104 or M140 (or G28) => collision of heads at end of print

SigmaFred, 2016-10-22:
I printed the Calibration Target (dual color) with the sigma. gcode was generated with Simply3D.
At the very end of the print (left head) the head was sent towards the parking position of the right head = collision.

I checked the end of the gcode and all seemed fine
`...`
`M104 S0 T0 ; left extruder off`
`M104 S0 T1 ; right extruder off`
`M140 S0 ; heated bed off`
`G28 X0 Y0 ; move XY to min endstops`
`M84 ; steppers off`

As the G28 code worked fine in the rest of the code - I just changed the line position 

And this sequence worked fine ...
`...`
`G28 X0 Y0 ; move XY to min endstops`
`M104 S0 T0 ; left extruder off`
`M104 S0 T1 ; right extruder off`
`M140 S0 ; heated bed off`
`M84 ; steppers off`

So I would assume that calling M104 or M140 might have an unwanted collateral effect on global variables used by G28. So at least one of the commands might be not properly implemented. 

Thank you.

chimpusmaximus, 2016-10-23:
Same issue here and heard others having same

AlejandroGarcia92, 2016-10-25:
I have opened a new firmware release candidate in order to fix this kind of issues before an official release. You can find the RC version here https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.2RC , so send us feedback.

closed 2017-04-05 by AlejandroGarcia92