BCN3D/BCN3DSigma-Firmware #104: There is a problem of compatibility with new firmware simpify3d.

ggangjuk, 2016-10-23:
1.2.0, 1.2.1, all versions of the firmware, there is a serious problem.

There is a problem of compatibility with new firmware simpify3d.

At the end of the print nozzles and moves in a strange position.

Firmware Version 1.2.0 is a test video link and gcode.
video :      https://drive.google.com/open?id=0B5YOgS9_4nxUdFU4UkQ5a3Bhb0E
gcode :      https://drive.google.com/open?id=0B5YOgS9_4nxURlZRNkcyeWRHeFE

Firmware Version 1.2.1 is a test video link and gcode.
video :      https://drive.google.com/open?id=0B5YOgS9_4nxUVDNCOW5sdUtOWDA
gcode :      https://drive.google.com/open?id=0B5YOgS9_4nxUSVJZOWtsOV9UZHc

The firmware version 1.6.1 is not a problem.

Please solve.

chimpusmaximus, 2016-10-23:
Same for me

SigmaFred, 2016-10-23:
Hi, 
I found a workaround at least for 1.2.0 (see post #103 )
  
In Simplify3D - Edit Process Settings - Scripts - Ending Script => I just changed the line sequence a bit (put the G28 to the top):

`G28 X0 Y0 ; move XY to min endstops`
`M104 S0 T0 ; left extruder off`
`M104 S0 T1 ; right extruder off`
`M140 S0 ; heated bed off`
`M84 ; steppers off`

This now works fine - but anyhow there is a bug.

AlejandroGarcia92, 2016-10-25:
I have opened a new firmware release candidate in order to fix this kind of issues before an official release. You can find the RC version here https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.2RC , so send us feedback.

tsp666, 2016-11-13:
Hi, your beta version is fixing the problem with simplify3d software generated code.

closed 2017-02-14 by AlejandroGarcia92