BCN3D/BCN3DSigma-Firmware #105: Minor Bug: FW get stuck when starting "bed leveling" after unload/load filament

mrcg, 2016-11-25:
FW is hanging if bed leveling feature is started after loading filament.
The issue was observerved with v1.2.0, but is existing since longer time.
Issue is 100% reproducable like this:
1. Unload filament
2. load filament
3. start "bed leveling"
The printbed is moving up and then FW is getting stuck.
The issue can only be resolved by power off/on.

It's quite usual to start bed leveling after filament change and therfore it's quite annoying.

Zorgonaute84, 2016-12-05:
It's a bug for sure, but honestly, I don't think that perform a bed leveling after a filament change is really necessary. I do not and I never had any troubles.

mrcg, 2016-12-07:
It's a minor bug existing since long time.
Anyhow it should be fixed and that's the reason why i am reporting this here.

I am personally performing bed leveling always after filament change.
Frist of all i think it's useful because of potential mechanical influence while changing the filament.
Secondly because it's the easiest way to move up the bed after filament change.

guillemavila, 2016-12-16:
This bug is fixed in the next RC we're publishing next week. We couldn't make it for today because it depends on some Graphical features too.

As @Zorgonaute84 says, it's not really necessary. Anyways by now you can move Z using the feature 'Move Z', that's faster than executing a bed calibration.

Thanks for the feedback!

mrcg, 2017-01-16:
Is the correction already in the latest 1.2.2RC?
It's not listed in the correction list and the 1.2.2RC which i have loaded (might not be the latest one) still has the issue.

AlejandroGarcia92, 2020-07-21:
This bug should be fixed on the current fw version.

closed 2020-07-21 by AlejandroGarcia92