BCN3D/BCN3DSigma-Firmware #108: Updated to 1.2.2RC and now printer bricked

maalencar, 2016-12-23:
Hello,

I updated my Sigma to v1.2.2RC. Updated firmware using Cura, reformatted MicroSD (FAT16) and installed LCD files for version 1.2.2RC. Restarted the printer. Logo animation took forever and printer eventually stays stuck on a screen that says "Welcome to the BCN3D Sigma experience". Nothing changes. touching the screen won't do anything. I re-installed everything two more times... same results. Please help resolve this as I basically have a large, useless metal box now.

Thanks

Mauro

AlejandroGarcia92, 2016-12-23:
Hello @maalencar,

I have updated the version RC. Try to download the version from this link: https://github.com/BCN3D/BCN3DSigma-Firmware/archive/v01-1.2.2RC.zip

Install manually the firmware using Cura Machine/Install Custom Firmware and copy the SD files again. If it doesn't work, you install the last official version(maybe you must do it manually as well).

Sorry for that but this branch is for test version.

kahunalux, 2017-01-18:
Mine is bricked when i try to install every versions since the 1.2  ....

Is that normal?

kahunalux, 2017-01-18:
I have fund the solution.
Don't copy on the sd with a mac or a linux ........

JimBarrows, 2017-02-10:
I am now in the same boat.  I tried the link above, and I have the same result.  This is all 100% windows, so I no longer have the copying from a mac issues I did have.

AlejandroGarcia92, 2017-03-02:
If you have reformatted to FAT16 and it doesn't work, there is a problem with this SD card.

tropicdome, 2017-03-02:
There has been no problems with the SD card before the firmware update, why would the firmware update make the SD card not work?

I downgraded to 1.2.1 and now everything works just fine again..

kitusmark, 2017-03-02:
Please check this little guide and see if it works 👍 

https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Updating-the-SD-Files-from-the-LCD-Display

tropicdome, 2017-03-02:
Hi @kitusmark,

Thank you for that, but that was exactly what I did, and it only worked for 1.2.1.

tsp666, 2017-03-02:
Hi, I can confirm this problem. the 1.2.2 firmware is running fine, but the lcd files do NOT. Slow logo and then white screen of death. running the same new firmware with the OLD 1.2.1 LCD files does work perfectly - just very little graphics like the additional forward in filament load are blanked  out.
So it 100% NOT about the SD card, but the content of the LCD files.

AlejandroGarcia92, 2017-03-03:
Can @tsp666 and @tropicdome tell us if your SD card is a Kingston Class 4? In this case contact with SAT and they will give you a solution.

tsp666, 2017-03-03:
Hi Alejandro,

yes, it's a CAT 4 micro SD - is this the problem, does it have to be a higher CAT? If so, which at least?
What do you mean by SAT ?

Regards
Torsten

AlejandroGarcia92, 2017-03-03:
Technical support

tsp666, 2017-03-03:
if it's just the card, so a replacement with a CAT 10 card would fix the problem?

tropicdome, 2017-03-03:
I will have to check when my print is done, but it is the card that came with the printer.

AlejandroGarcia92, 2017-03-03:
The problem is not in the CAT 4, is in the Kington CAT 4. I believe if you try with a CAT 10 should work properly. But using a SD card CAT 10 you can not format to FAT16 easy with windows, you should follow that: https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Updating-the-SD-Files-from-the-LCD-Display

tsp666, 2017-03-03:
even an older CAT 1 or 2 micro 2 GB card did the trick, it's working now with the 1.2.2 LCD files

kikolobo, 2017-03-06:
I can confirm this issue happens even when following instructions to the detail on Windows. 

Tried this steps:  https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Updating-the-SD-Files-from-the-LCD-Display. 

Didn't work. I tried downgrading firmware and SD files... Didn't work. My guess this is a firmware issue more than an SD card issue. Maybe some fuse bits changed or SPI speed or something? Will check the diff.

AlejandroGarcia92, 2017-03-06:
@kikolobo can you confirm us if your LCD SD Card is a Kingston CAT 4?

kikolobo, 2017-03-08:
Yes. Confirmed.. It is.

Si es 😊 Soy de Mexico 😊 No sabia que eran de España! Felicidades por la maquina.

eephrati, 2017-04-02:
Hi there,

I had the  same problem and ended up in this page. When i did the firmware upgrade and updated the LCD files, the startup was really slow with the logo and it would just stay in that page. I tried different SD Card but turned out that it is a SD Speed issue. 

CAT 4 is too slow. I bought a CAT 10 and worked beautifully. 

The only issue is that it needs to be in FAT16 format, not FAT 32 (and in order to get it FAT16, it needs to have a 4gb memory or less). I had a hard time finding out 4GB SD Card.

So I bought 8GB and created a 4GB Partition in order to make the file System at FAT16. Here is a good instructable on how to do it on Windows.
(http://www.instructables.com/id/Format-USB-Flash-Drive-to-FATFAT16-not-FAT32/?ALLSTEPS)

After copying the new file on the new FAT16, CAT10, 4gb partition, I plugged it into the 3D Printer (Sigma) and it worked perfectly.

gjweppler, 2017-06-06:
Same here upon upgrading from 1.2.2 to 1.2.4. Used Kingston 4GB uSD and created 500MB FAT16 partition following the guide to the letter. Tried with different uSD cards (cat4, cat10) and with multiple different tools to format from two different PCs (win7 & win10). Still ending up at the slow logo display @power-on. Oh the pain...

closed 2017-09-19 by AlejandroGarcia92