BCN3D/BCN3DSigma-Firmware #111: Fix compilation on Linux

Wnt, 2017-02-03:
Compilation fails on Linux with message:
`SD_ListFiles.h:7:22: fatal error: arduino.h: No such file or directory   #include "arduino.h"`
the file name is Arduino.h, not arduino.h. This probably also affects OS X users.

Wnt, 2017-07-18:
Why the 'wontfix' tag? I find it pretty rude to have that added without any comment to explain it. After all the fix is just changing one character to uppercase. A change that should not affect Windows developers at all.

AlejandroGarcia92, 2017-07-18:
@Wnt Apologies. This line works on Windows complitation but the file is Arduino.h as you said. I realized that this include is not necesary so this line can be remove as you wish or changed to #include "Arduino.h". I will change it at the master branch.

AlejandroGarcia92, 2017-07-18:
[Commit](https://github.com/BCN3D/BCN3DSigma-Firmware/commit/d65e172ca2f1097acdb190b05d3707488b1e8e3f)

closed 2017-07-18 by AlejandroGarcia92